unihd-cag/skillbridge #56: Setting a logLevel from Virtuoso results in a not responding Server

TM90, 2019-11-06:
pyStartServer with any logLevel even "WARNING" results in the server not responding
pyStartServer without the logLevel argument works.

nielsbuwen, 2019-11-06:
The logLevel "DEBUG" definitely works:

```
pyStartServer "default" "DEBUG"
```

TM90, 2019-11-06:
Using it like this in the skill console produces the error pyStartServer(logLevel="DEBUG")

nielsbuwen, 2019-11-06:
Looks like you cannot use optional parameters like this

```
defun(test (@optional (x 1) (y 2))
    printf("x=%L y=%L\n" x y)
)

test(y=20)
```

Prints "x=20 y=2"

TM90, 2019-11-06:
Ok then we should update the documentation.

TM90, 2019-11-06:
@key has the intended behaviour:

```
defun(test (@key (x 1) (y 2))
	printf("x=%L y=%L\n" x y)
)

test
>>> x=1 y=2
test(?y 10)
>>> x=1 y=10
```

nielsbuwen, 2019-11-07:
Closed with #57

closed 2019-11-07 by nielsbuwen