unihd-cag/skillbridge #59: pyStartServer exits with error code 1

iamanintrovert, 2019-11-08:
I used `pip install skillbirdge`. Then in Virtuoso CIW ran `load("path\to\python_server.il")` and `pyDumpFunctionDefinitions "<install>"`.
After that on the same CIW, when I run `pyStartServer` I get
`error "python: can't open file '/path/to/lib/python3.6/site-packages/skillbridge/server/python_server.ipy': [Errno 2] No such file or directory\n"
server was stopped with code 2`

I thought I needed to close Virtuoso and run the two commands again. But when I did that I got this.
`error "  File \"/path/to/lib/python3.6/site-packages/skillbridge/server/python_server.py\", line 21\n    def send_to_skill(data: str) -> None:\n                          ^\nSyntaxError: invalid syntax\n"
server was stopped with code 1`

Am I missing anything? Isn't "Open the Skill console in Virtuoso and type" the same thing as "typing in Virtuoso CIW" ?
python version 3.6
virtuoso 6.1.7

TM90, 2019-11-08:
You have to type it in the CIW that is correct.

When you run `python `on your system does it run the interpreter for python 3.6 
and  `python -m skillbridge ` yields the path?

nielsbuwen, 2019-11-08:
The first error looks rather strange: `python_server.ipy` really doesn't exist, because it is called `python_server.py` (without the i). That sometimes happens when you accidentally include a space in the `load` command: `load("path-to-python_server.il               ").

We should change the way we handle paths in Skill (#60).

The second error happens, because an older python version was used when starting the python server. Maybe we can also improve that (#61).

In the meantime, please try the following **workaround**:

1. identify the newest python version (let's call it `python3.6`)
2. create a virtual environment `python3.6 -m venv env`
3. activate the environment `source env/bin/activate`
4. install the skillbridge inside the environment `pip install skillbridge` (it should reinstall)
5. find out the path to the server `python -m skillbridge` (this time **don't** use python3.6)
6. start Virtuoso inside the environment `virtuoso&`
7. load the server script `load("new-path-to-server.il")` and please make absolutely sure that there are no spaces before or after the path.
8. dump the function definitions again `pyDumpFunctionDefinitons "<install>"`
9. start the server `pyStartServer`

When you use it next time you only need to do steps: 3., 6., 7., 9.

iamanintrovert, 2019-11-08:
Thanks for the instructions. I think the problem was I did not start Virtuoso from the python virtual environment. Once I started Virtuoso by running the startscript from the python virtual environment `pyStartServer` worked fine.

closed 2019-11-08 by iamanintrovert

daquintero, 2021-03-09:
Hi! Many thanks for all your amazing work on this.

I can't find out what I'm doing wrong and would appreciate some guidance. I think I've followed the instructions above to the letter, but when I run the above 8. and 9., I get the following traceback:

`pyDumpFunctionDefinitons "<install>"
*Error* eval: undefined function - pyDumpFunctionDefinitons
pyStartServer
t
ERROR python: symbol lookup error: python: undefined symbol: _Py_LegacyLocaleDetected

server was stopped with code 127`

I have the impression it has to do with my virtual environment setup. Does this ring any bells for you on what it could be? I Will update as I am debugging.

TM90, 2021-03-09:
What is your used python version? Make sure python maps to a python interpreter Version 3.6 or higher.

nielsbuwen, 2021-03-09:
Hi, thank you for your interest in our project.

This error does indeed ring a bell. We have encountered it in certain setups but we are not sure what exactly is causing it.

Fortunately there is a workaround. It may interfere with your setup, however.

Instead of `pyStartServer` try `pyStartServer ?python "LD_LIBRARY_PATH= python"` (the space after the equals sign is important). This tells the shell to clear the `LD_LIBRARY_PATH` variable before starting python.

TM90, 2021-03-09:
Also pyDumpFunctionDefinitons "<install>" is not needed anymore.