unihd-cag/skillbridge #62: how to make OCEAN functions availble for use in python?

iamanintrovert, 2019-11-09:
I was trying to get OCEAN functions like `simulator`, `modelFile`, `resultsDir` etc available to python through skillbridge. I modified the `listFunctions` function argument as `"[a-z]"` in the `python_server.il` file so that it includes every kind of SKILL functions when I run `pyDumpFunctionDefinitions`. 

However I don't know how to use those the newly added OCEAN functions (like `simulator`). 
I could not get those functions through `ws = Workspace.open()` then `ws.simulator`.

Does skillbridge has the functionality to use OCEAN functions?

TM90, 2019-11-09:
Hello,

The skillbridge basically maps everything you can do in the CIW. Ocean is in my understanding a subsystem of Virtuoso. 

Nevertheless there for every subsystem in Virtuoso there is a skill interface usable from the "CIW"
These interfaces normally have a prefix. In example the database Interface has the prefix db. A prominent example would be the skill Function `dbOpenCellView("lib", "cell_name", "view_name")` which maps with the skillbridge to `ws.db.open_cell_view_by_type("lib", "cell_name", "view_name")`.

There are also skill interfaces for the simulators (`ws.axl.<function>`) and an ocean interface (`ws.ocnxl.<function>`).

If you are using jupyter-notebook or an interactive python interpreter you can use the tab completion to complete function names.

Otherwise, you can use `dir(ws)` to list all prefixes. and `dir(ws.ocnxl)` to list all functions within. If you then run a function without an argument you will get the skill doc string back.

```
>>>ws.ocnxl.begin_test
<remote function 'ocnxlBeginTest'>
ocnxlBeginTest(
t_testName
)
=> t / nil

This command indicates the beginning of the test specified by
testName. Subsequent commands populate this test. The test specification
ends when ocnxlEndTest() command is given.
 ```

It is really helpful for such a task to use the cadence online support or the cadence help (`cdnshelp` on your command line). Each skill interface should be documented there.

I hope this helps.

Best regards

iamanintrovert, 2019-11-10:
Thanks for the explanation. Basically I am trying to run spectre simulation from python and I have somewhat familiarity of doing the sim using ocean. Sp it would have been nice if I could access the ocean functions. A workaround could be to implement a SKILL function where I use the ocean functions and register it in skillbridge.

TM90, 2019-11-10:
Hello,

If you already have a lot of ocean code this would be a working solution.

If you just want to start a simulation for which you have a maestro view.

you could use the ADE Explorer/Assembler Skill interface which is documented in the Virtuoso ADE Explorer and ADE Assembler SKILL Reference which can be found in the cdnshelp or cadence online support.

`ws.mae.open_setup` and `ws.mae.run_simulation` look promising.

nielsbuwen, 2019-11-10:
As a last resort you can indeed register Skill functions you wrote:

```
from skillbridge import Workspace
from skillbridge.client.functions import Function

ws = Workspace.open()
ws.user += Function("myFunction", "documentation of myFunction", set())

ws.user.my_function(...)
```

TM90, 2019-11-10:
> 
> You can even use this method to map existing functions like `simulator`.

`simulator` is not an existing function in the CIW context. But if you can call it from there you can wrap it and map the wrapper function.

iamanintrovert, 2019-11-10:
> As a last resort you can indeed register Skill functions you wrote:
> 
> ```
> from skillbridge import Workspace
> from skillbridge.client.functions import Function
> 
> ws = Workspace.open()
> ws.user += Function("myFunction", "documentation of myFunction", set())
> 
> ws.user.my_function(...)
> ```
> 
> You can even use this method to map existing functions like `simulator`.

Thanks! This is indeed what I needed. I am able to call my custom skill functions from python.

closed 2019-11-14 by nielsbuwen