unihd-cag/skillbridge #76: pyStartServer ?timeout nil returns error

iamanintrovert, 2019-11-19:
I was seeing a timeout error stay connected to the server and run some operations for a few hours. So I decided to set a longer timeout period by `pyStartServer ?timeout nil`. But virtuoso returns this error

`*Error* ilGetString: arg must be symbol or string - nil`

I tried the command mentioned in the documentation `pyStartServer ?timeout 10.5` which also returns the same error. 

`*Error* ilGetString: arg must be symbol or string - 10.5`

sending `10.5` as string `"10.5"` returns

`pyStartServer ?timeout "10.5"
t
error "usage: /......./lib/python3.6/site-packages/skillbridge/server/python_server.py\n       [-h] [--notify] [--single]\n       socket {DEBUG,WARNING,INFO,ERROR,CRITICAL,FATAL}\n/tmp/mhasan13-lws54/anaconda3/envs/brian/lib/python3.6/site-packages/skillbridge/server/python_server.py: error: argument log_level: invalid choice: '10.5' (choose from 'DEBUG', 'WARNING', 'INFO', 'ERROR', 'CRITICAL', 'FATAL')\n"
server was stopped with code 2`

What is the correct way to set timeout to some value or nil?

TM90, 2019-11-19:
Which version are you running? The timeout feature is quite new.

To see which version you have installed run:

`pip show skillbridge`

to update run:

`pip install skillbridge --upgrade`

iamanintrovert, 2019-11-19:
I was running version `1.0.2`
After I upgraded to version `1.0.4` timeout feature is working as expected. Thanks!

closed 2019-11-19 by TM90