unihd-cag/skillbridge #114: pyStartServer not working

avlsi, 2020-02-20:
Hi,

After doing load("python_server.il")
Unable to do pyStartserver

ERROR ", line 21

ERROR     def send_to_skill(data: str) -> None:

ERROR                           ^

ERROR SyntaxError: invalid syntax

Virtuoso 6.1.7-64b is used.

TM90, 2020-02-20:
Did you install the package with pip ?
Afterwards you use` skillbridge path` in the terminal to get the correct path ?

The `load("your/path/python_server.il") `in the virtuoso CIW
Afterwards run `pyStartServer()`

Which Python version are you using and please make sure that virtuoso runs the correct python interpreter. 

The recommended approach is to setup a python virtual environment and then start Virtuoso in your started environment where the python alias maps to python >=3.6

avlsi, 2020-02-20:
Thanks after updating the custom interpreter path, the command is returning 't' in CIW.
I am able to  do pyDumpFunctionDefinitions "<install>" also.
But the example to check skill function is not working when I am running from a linux terminal.
- code in check_skillbridge.py -
from skillbridge import Workspace
ws = Workspace.open('foo')
cell_view = ws.ge.get_edit_cell_view()

print(dir(cell_view))
print(cell_view.b_box)
------------------------------------------------
I get the following error
bash % python ../work/check_skillbridge.py 
Traceback (most recent call last):
  File "../work/check_skillbridge.py", line 6, in <module>
    ws = Workspace.open('foo')

File "/home/avlsi/local/lib/python3.6/site-packages/skillbridge/client/workspace.py", line 238, in open
    raise RuntimeError("No server found. Is it running?") from None
RuntimeError: No server found. Is it running?

TM90, 2020-02-20:
Try `ws = Workspace.open()` The argument is used to connect with an ID when you want to access multiple virtuoso instances on one machine .

To verify if the server is actually running run `ps` in your terminal. There should be the following processes:
- cdsServIpc
- python

avlsi, 2020-02-20:
I think I understand the issue. 
I am launching the virtuoso on a server. I need to figure out how to access the terminal where the processes are running.

TM90, 2020-02-20:
Yes that sounds that this would be the issue. 
I will close this issue. 

If there are other problems feel free to open another one.

Lshnn, 2023-08-03:
Hi, I'm facing exactly the same issue : raise RuntimeError("No server found. Is it running?") from None
RuntimeError: No server found. Is it running?

I would like to know how did you resolve it. Thanks

nielsbuwen, 2023-08-03:
Did you run `pyStartServer` before connecting with `Workspace.open()`?

Lshnn, 2023-08-03:
Yes I ran the pyStartServer on CIW before.

nielsbuwen, 2023-08-03:
Can you share some more information please.

- Python version
- Skilbridge version
- The Cadence Tool you are using
- How you started the server
- The python code that contains the Workspace

Lshnn, 2023-08-03:
Python version 3.8.2; skillbridge-1.5.0; and virtuoso 6.1.8
I started the server like described in the documentation : I ran the skillbridge path on a terminal then I loaded the path to "python_server.il" to in the CIW after that I ran the pyStartServer ?python "python3.8".
once I didi that, on a shell terminal I run the python3.8 and run the two python lines :
from skillbridge import Workspace
ws = Workspace.open()

nielsbuwen, 2023-08-03:
Can you please verify that a file is created when you run `pyStartServer`. It is called `/tmp/skill-server-default.sock`.

Lshnn, 2023-08-03:
I checked my /tmp and can't see this file

nielsbuwen, 2023-08-03:
Maybe you don't have enough permissions to create and read files in `/tmp` .

Does the following work for you:

```bash
touch /tmp/skillbridge-test
cat /tmp/skillbridge-test
rm /tmp/skillbirdge-test
```

nielsbuwen, 2023-08-04:
Unfortunately i'm running out of ideas, here :(

Did virtuoso create log files in the directory where you started it? If yes what is inside, if no i don't know what's the problem. Sorry

Lshnn, 2023-08-04:
yes it created skillbridge_server.log file containing this :

02.08.2023 08:21:57 INFO starting server id=default log=INFO notify=False single=False timeout=None
02.08.2023 09:09:32 INFO starting server id=foo log=INFO notify=False single=False timeout=None
02.08.2023 09:32:44 INFO starting server id=default log=INFO notify=False single=False timeout=None
03.08.2023 02:30:41 INFO starting server id=default log=INFO notify=False single=False timeout=None

jesseyu1984918, 2025-02-22:
Looks like python 3.11 will have the problem as well and I can't use 3.6 due to other package.

TM90, 2025-02-24:
> Looks like python 3.11 will have the problem as well and I can't use 3.6 due to other package.

All python versions above 3.8 should work. Can you describe what you tried so we can find the underlying issue here?

closed 2025-03-04 by TM90