unihd-cag/skillbridge #115: Guide for loading user defined skill function

avlsi, 2020-02-20:
Is there a tutorial on how to use user defined skill functions from a user.il file?

Reference - https://github.com/unihd-cag/skillbridge/blob/199c083623b73a0b4a00c887b83627a284060d54/docs/examples/custom_functions.rst

avlsi, 2020-02-21:
skill.il has a function to return a list of cells in the given library
the procedure is called get_cells_in_library

I added the following code to test the custom functions from a skill file.

from skillbridge import Workspace
ws = Workspace.open()
@ws.register
def get_cells_in_library(lib_name) -> None:
    """
    return a list of cells in the given library
    """
library_name = "check_lib"
print(get_cells_in_library(library_name))

I get the following error
TypeError: 'Function' object is not callable

TM90, 2020-02-21:
@ws.register is used to register a function based on our prefix approach.

In example if you have a skill method called `userGetLibs`

You register it with:

```
@ws.register
def userGetLibs(lib_name) -> "Your return annotation":
    """
    return a list of cells in the given library
    """
```

Afterwards you can acess the function in python with:

`ws.user.get_libs()`

Alternatively if you want to map a function without the prefix approach take a look at this issue #62 :

Here is the important part from that issue: 
You can manually categorize and register functions without the auto prefix categorization:

```
from skillbridge import Workspace
from skillbridge.client.functions import Function

ws = Workspace.open()
ws.user += Function("myFunction", "documentation of myFunction", set())

ws.user.my_function(...)
```
i.e.:
```

from skillbridge import Workspace
from skillbridge.client.functions import Function

ws = Workspace.open()
ws.user += Function("get_cells_in_library","returns a list of cells in the given library", set())
ws.user.get_cells_in_library()
```

avlsi, 2020-02-22:
I tried with couple of userdefined procedures and its able to work with that.

avlsi, 2020-02-22:
One question though, why can't there be a wrapper inside this skillbridge where in user can run a command of his choice after starting server.
like -> load("new.il") and get those procedures automatically

TM90, 2020-02-24:
We did not see registering a lot of functions from a skill file inside the scope for the skillbridge.
Especially since it can be implemented on top.

All you have todo is to implement the following Functionality:

1.  map the skill load function
2. load new.il
```
from skillbridge import Workspace
from skillbridge.client.functions import Function

ws = Workspace.open()
ws.user += Function("load","loads skill code inside the virtuoso env", set())
ws.user.load("new.il")
```
3. read in new.il in python parse the file line by line and detect skill function definitions
4. parse the function name from the definition
5. register the function with the name parsed and add a generic documentation

Would it help you if there was a functionality which maps user functions with a specific prefix like "userSomeFunctoin" automatically? That would be a feature which could be implemented in a way that you can load these functions and then run `pyDumpFunctions` inside the CIW to map them automatically to the user prefix

rdesaiTDK, 2020-03-18:
Does this solution still apply?
[image]

Trying to run a bunch of user-defined skill functions. What is the efficient way to do it?

nielsbuwen, 2020-03-19:
@rdesaiTDK 

If your function follows the SKILL camel case naming convention, you may use the recommended approach and register it:

```
@ws.register
def prefix_my_function_name(args_here) -> return_annotation_here:
    """docstring here"""
```

Then you can call it like this:

```
ws.prefix.my_function_name(...)
```

At the moment it is impossible with the public API to call functions that deviate from this naming scheme. Let me know, if you need an alternative approach.

TM90, 2020-03-19:
With skillbridge 1.2.3 we now have an official documented way to directly call skill methods without registering them:

Release:
https://github.com/unihd-cag/skillbridge/releases/tag/releases%2F1.2.3

Update with:

pip install -U skillbridge

Documentation:

https://unihd-cag.github.io/skillbridge/examples/custom_functions.html

TM90, 2020-03-20:
@rdesaiTDK Does this solve your problem?

rdesaiTDK, 2020-03-21:
Thanks for the quick help and sorry for the late response @TM90.
Happy to report, the Direct Method of calling Skill Methods works perfectly in Rel 1.2.3.

Kudos to you all for developing this. 

Stay Home & Stay Safe, 
Ronak

TM90, 2020-03-21:
Thank you for the feedback!

closed 2020-03-21 by TM90

curtisma, 2020-03-21:
Does skillbridge support asynchronous calls to SKILL?

I'm trying to make a comparison to the SKILL-Python server design that BAG
uses which does allow asynchronous calls.

Regards,
Curtis

avlsi, 2020-03-22:
I tried the same after updating the skillbridge in my local installation
I am getting following error when server start is launched
pyStartServer
t
ERROR   File "/home/$USER/.local/lib/python3.6/site-packages/skillbridge/server/python_server.py", line 21
    def send_to_skill(data: str) -> None:
           
ERROR                ^
SyntaxError: invalid syntax

TM90, 2020-03-22:
There seems to be something wrong with your python environment.

In #114 and #59 are snippets on how to debug the environment and some best practices.

avlsi, 2020-03-22:
Its working. I was able to load the custom function using the new definition format.

PR #125 (2020-03-19): Added __get_item__ to Workspace for direct functions
closes #115