unihd-cag/skillbridge #126: Does skillbridge support asynchronous calls to SKILL?

TM90, 2020-03-21:
Does skillbridge support asynchronous calls to SKILL?

I'm trying to make a comparison to the SKILL-Python server design that BAG
uses which does allow asynchronous calls.

Regards,
Curtis

On Sat, Mar 21, 2020 at 6:38 AM Tobias Markus <notifications@github.com>
wrote:

> Closed #115 <https://github.com/unihd-cag/skillbridge/issues/115>.
>
> —
> You are receiving this because you are subscribed to this thread.
> Reply to this email directly, view it on GitHub
> <https://github.com/unihd-cag/skillbridge/issues/115#event-3151669136>,
> or unsubscribe
> <https://github.com/notifications/unsubscribe-auth/ABDX4SKTLEPRV772NSIOEADRISRKTANCNFSM4KYUAFPQ>
> .
>

-- 
Curtis Mayberry
Analog and RF Design Engineer
www.CurtisMayberry.com
515-451-5625

_Originally posted by @curtisma in https://github.com/unihd-cag/skillbridge/issues/115#issuecomment-602107640_

TM90, 2020-03-21:
In my opinion, it is not in the scope of the skillbridge to provide an asynchronous interface. We just provide a slim layer for language mapping.

In my opinion, python, since 3.5 should with asyncio, has everything to create asynchronous calls.

I will try to give you an example on Monday.

TM90, 2020-03-23:
I was wrong asyncio by itself does not offer a solution.
The skillbridge is fundamentally synchronous since the Virtuoso skill interpreter is synchronous.

Could you give us a use case for asynchronous calls?

If you want for example start a long simulation and calculate in parallel in python this can be done with the Thread package.

``` python
from skillbrige import Workspace
from threading import Thread

ws = Workspace.open()

def pre_calculate()
    pass

thread = Thread(target=pre_calculate)
thread.start()
ws.simulate()
```

Skillbride commands should always be within one thread.

closed 2020-03-26 by TM90

curtisma, 2020-03-27:
Could you give us a use case for asynchronous calls?

You could start several simulations and have them run in parallel,
especially if you're using LSF.

Another case is if you need to run DRC, LVS, or parasitic extraction on a
design you could have them both run at the same time while the python code
is also running.

TM90, 2020-03-27:
> Could you give us a use case for asynchronous calls? You could start several simulations and have them run in parallel, especially if you're using LSF. Another case is if you need to run DRC, LVS, or parasitic extraction on a design you could have them both run at the same time while the python code is also running.
> […](#)
> -- Curtis Mayberry Analog and RF Design Engineer [www.CurtisMayberry.com](http://www.CurtisMayberry.com) 515-451-5625

Running multiple Simulations and DRC runs can be setup via the skill  interface of the simulator or the drc tool and thus via the skillbridge.

The example given should solve running python in parallel to a long running skillbridge task.

Best regards

Tobias

curtisma, 2020-03-28:
Ah, I missed the example you gave previously since my email client hid it,
but I see it now.

Thanks,
Curtis