unihd-cag/skillbridge #127: How to check for exceptions when using this skillbridge

avlsi, 2020-03-22:
I am writing a basic wrapper to use skillbridge for my purpose.
from skillbridge import Workspace
class virtuoso(object):
    """
    check whether pyStartServer is running before doing this
    """
    def __init__(self):
        ## raiseerror here if server is not running
        self.ws = Workspace.open()

    def load_custom_skill_files(self,filelist):
        self.customloadfiles = filelist
        for file in self.customloadfiles:
            self.ws['load'](file)

    def create_schematic(self,libname,cellname):
        """
        """
        self.libname = libname
        self.cellname = cellname
        self.ws.db.open_cell_view_by_type(self.libname, self.cellname,"schematic","schematic","w")

class skillCode(virtuoso):
      def __init__(self):
        super(skillCode, self).__init__()

      def get_symbol_pins(self,libname,cellname):
            pass
I would like to know how to add exception handling like checking whether library exists or cellname exists and also is there a better way to do this wrapper?

nielsbuwen, 2020-03-23:
There seem to be two questions:

1. How can you check if the Workspace is connected to Virtuoso
2. How to handle the errors that occur when calling SKILL functions

`1.` This is a little bit of a problem. If you never started the SKILL server before or if you shut down the SKILL server properly last time, a `RuntimeError` is raised in case the connection failed:

```python
from skillbridge import Workspace

# if the server was shut down properly and is not running currently
try:
    ws = Workspace.open()
except RuntimeError:
    print("Server is not running")
```

But sometimes the socket file survives even thought the server was closed. In that case the error will only occur when you call a SKILL function. It can either be a `ConnectionRefusedError` or a `BrokenPipeError`.

```python
from skillbridge import Workspace

# if the server was killed
ws = Workspace.open()  # no error
try:
    ws.ge.get_edit_cell_view()
except (ConnectionRefusedError, BrokenPipeError):
    print("Server is not running")
```

We are a bit out of ideas on how to fix that. The core problem is that we don't know how to determin if a socket file is still valid. We can only try to send messages and see if that fails.

`2.` The error handling for arbitrary SKILL functions depends on the SKILL function. Some just return `None` if they fail and some actually raise a SKILL Error. These errors are translated into `RuntimeError`. Sometimes the SKILL functions print warnings into the CIW, these are also translated into python as `UserWarning`.

```python
# if no cell view is open
cv = ws.ge.get_edit_cell_view()  # None

# wrong number of arguments
try:
    ws.ge.get_edit_cell_view(1, 2, 3)
except RuntimeError:
    print("An error occured")
```

You must check the SKILL reference for the specific behaviour of the function you are calling.

TM90, 2020-03-27:
This seems to be resolved. If there are follow up questions feel free to open this issue again.

closed 2020-03-27 by TM90

curtisma, 2020-03-28:
For #1:
Couldn't you do a handshake where skillbridge calls a skill function and
expects a certain reply.  You could just run that when opening the
workspace to check if the connection is working correctly and the SKILL
server is running.

-Curtis

nielsbuwen, 2020-04-01:
The handshake is a good idea. For now we won't include it, because it can be done very easily with the new features from #124.

A simple handshake example

```python
from skillbridge import Workspace

try:
    ws = Workspace.open()
    assert ws['length']([]) == 0  # the length of an empty list should be 0
except (ConnectionRefusedError, ValueError, BrokenPipeError, AssertionError):
    print("Error")
else:
    print("Connected")
```