unihd-cag/skillbridge #140: RemoteObject Fails to parse 'stdobj@0xHEXNUM' type data from skill

iamanintrovert, 2020-05-09:
skill functions like `paramAnalysis`, `selectResult` returns `stdobj@0xHEXNUM` type object which skillbridge is not able to recognize. When I use these skill functions using `ws['paramAnalysis'](...)` it issues the following error in CIW. 

`command "__py_stdobj_0x2a183c20->objType" resulted in error ("slotValue" 0 t nil (*Error* slotValue: no such slot - objType in class apaParametricAnalysis. Valid slot names: (session analysisList parent psl original sweepName rangeList remainingRangeList rootDir rootPsfDir tmpDir sweepType)"))
`

The `paramAnalysis()` function runs fine,  because `RemoteFunction` `__call__` method's `result` variable shows a result. But it is still issuing error when trying to decode it. Other object types such as `srrWave@xHEXNUM` returned by `getData()`.

Is there a fix for this in the current version of skillbridge?

nielsbuwen, 2020-05-10:
Hi,

thank you for the report.

Just to be sure: the error occurs only when you print the result of the `paramAlanylsis` function in python, correct? So this code should not produce any error:

```python
result = ws['paramAnalysis'](...)
```

But printing the `result` does?

```python
print(result)  # error
str(result)  # error
repr(result)  # error
```

This is because we access the `objType` attribute everytime the string representation for a remote object is created. The obvious workaround is to never convert the object into a string.

I will try to reproduce this on monday and fix it.

iamanintrovert, 2020-05-10:
The error occurs on this line before the result can be printed.
```python
result = ws['paramAnalysis'](...)
```
The easiest to way to reproduce the error is to run a very simple (an NMOS drain current when gate and drain voltage varies) parametric sim in cadence, export the OCEAN script and start converting the skill functions in the OCEAN script to python.

Thank you for your effort on this great project.

iamanintrovert, 2020-05-10:
accidentally pressed close issue button. :)

TM90, 2020-05-11:
We both don't use ocean for simulation automation. So it is difficult to reproduce the error:
We created a test simulation under the library test with the cell name sim and the cell_view schematic.

And tried the following for us the command only fails if it is printed.

``` python
ws['simulator'](Symbol(spectre))
ws['design']("netlist")
ws['desVar']('test' 0.1)
x = ws['paramAnalysis'] ("test", start=0, stop=1, step=0.1)
```

Which only fails for us if printed. Do you may have a more specific test case, where we can reproduce the error on function call. An thus we can only fix the representation part.

iamanintrovert, 2020-05-11:
Hello,
In my case it is also failing when printed. The error occurs on function call when return value is not assigned to anything. That is when called like this 

```python
ws['paramAnalysis'](...)
ws['selectResult'](Symbol('dc')) # this also returns stdobj
```
which I think related to the same issue as printing the result.

TM90, 2020-05-13:
With Release 1.2.7 the issue should be resolved. Could you verify that it works in your case?

TM90, 2020-05-14:
I will close this issue for if it did not solve your problem feel free to reopen it.

closed 2020-05-14 by TM90

iamanintrovert, 2020-05-16:
> 
> 
> With Release 1.2.7 the issue should be resolved. Could you verify that it works in your case?
> https://github.com/unihd-cag/skillbridge/releases/tag/releases%2F1.2.7

The issue seems to be fixed. Thanks!

TM90, 2020-05-19:
We will add the CIW error issue to our issue list. We know what is causing it. But we don't have a clean fix for it yet.