unihd-cag/skillbridge #143: Error in create path example - ws.rod.create_path

JacobEFO, 2020-06-09:
Hi guys,

Excellent tool I found here. I am currently trying to automate some layout tasks and followed your example on [github.io](https://unihd-cag.github.io/skillbridge/examples/create_path.html).

However, when you call
`ws.rod.create_path(layer=["Metal2", "drawing"], width=0.08, pts = [[0. + 0.125 * n, 0.], [0. + 0.125 * n, 5.]])`
You seem to be missing the ?cdId argument (the database ID for the cellview in which you are creating a ROD path). This is documented on page 72 of the Virtuoso Relative Object Design SKILL Reference. 

To solve this, I alter the function to:
`ws.rod.create_path(cvId=cv, layer=["M1", "drawing"], width=0.08, pts=[[0. + 0.125 * n, 0.], [0. + 0.125 * n, 5.]])`
I am however puzzled why I need to call `cVid=cv` and not `cdId=cv`.

I hope this may be of help.

Thanks for your effort in reducing the need to do SKILL scripting ;)

nielsbuwen, 2020-06-09:
Hi,

thank you for your interest in our project and thanks for reporting this issue.

Seems like we have an error in our examples.

However the SKILL reference documents the cell view id parameter as `[ ?cvId d_cvId ]`. So the python variant of that should be called `cv_id=cv` (or `cvId=cv`). Maybe you have an outdated version of the reference with different parameter names.

> Thanks for your effort in reducing the need to do SKILL scripting ;)

that was our primary motivation ;)

JacobEFO, 2020-06-09:
Hi,

I see `[?cdId d_cvId]` from the documentation for product version ICADVM18.1 from January 2020 fresh from Cadence's online ressources today.

So I am not sure, if there's an error in my documentation or what happens. From the error I received, I got the hint to use `cvId` instead of `cdId` (as hinted from my documentation).

closed 2020-06-09 by TM90

TM90, 2020-06-09:
> Hi,
>
> I see `[?cdId d_cvId]` from the documentation for product version ICADVM18.1 from January 2020 fresh from Cadence's online ressources today.
>
> So I am not sure, if there's an error in my [...]

I guess that there is a small typo in the cadence documentation. The example should now be up to date with Release 1.2.8. Thank you.

PR #144 (2020-06-09): Fix error in example
This fixes an error in the "Create some paths" example. Also it removes some unnecessary code.

Closes #143