unihd-cag/skillbridge #145: Using leCreatePin to automatically generating pins & labels

JacobEFO, 2020-06-12:
Hello again ;)

I'm currently working on automatically generating layout pins, so I can easily visualize my layout floorplanning and interconnects. For this I expect to be using leCreatePin (documented in the Virtuoso Layout Suite Skill Reference). It has listed the following example:
`leCreatePin( geGetEditRep() "metal1R" "rectangle" list(0:0 10:10) "term1" "input"
list("left" "right") )`

However, I fail to realize how I can use the python list `[]` to correctly generate the `list(0:0 10:10)`. I know that I can use the `keys()` to have keyed arguments (in Skill) and use `[]` to create my list. But it's not entirely clear to me, how I map different syntaxes in Python correctly to Skill.

Can you point me towards some documentation, where I can figure this out?

nielsbuwen, 2020-06-12:
Hi,

The syntax construct `NUMBER:NUMBER` in SKILL simply is a shortcut for a list of two numbers. It's mainly used to denote 2D points. 

So in python it translates to `[NUMBER, NUMBER]`. In your specific example you need `[[0, 0], [10, 10]]`. The `keys()` function isn't neccessary here.

I hope this answers your question. We probably should add this to the example section in our documentation.

Feel free to close the issue.

JacobEFO, 2020-06-13:
Hi thanks for your assistance.

Okay, it makes sense that we can map 0:0 with the lists function. I also discovered an issue with calling it with keyword arguments. It seems, that I cannot use keywords like `cell_view_id=cv`, `layer_purpose_pair=["M1", "pin"]`.

How can I distinguish when I can use keyword arguments and not? I've gotten the hang of arguments such as `d_cellViewId`; 'd' refers to the type and is ignored and 'cellViewId' can be used as is or changed to 'cell_view_id' (although this seem to fail in the specific case).

For reference, I found the following example works and creates a pin:
`from skillbridge import Workspace
ws = Workspace.open()
cv = ws.ge.get_edit_cell_view()
ws.le.create_pin(cv, ["M1", "pin"], "rectangle", [
                 [0, 0], [10, 10]], "test", "input", ["top", "bottom", "left", "right"])
ws.hi.redraw()`

nielsbuwen, 2020-06-14:
The correct way to call an arbitrary SKILL function can only be obtained from the official SKILL documentation. We tried to unify the skillbridge interface but we were unable to identify a pattern in the documentation. That's why the user of the skillbridge library must lookup the function documentation before using it.

We can only provide a shortcut for looking up a short version of the documentation:

```python
print(ws.le.create_pin)
```

I am not sure how much of the official SKILL documentation i may post here, but i can show an example:

If the function documentation looks similar to this:

```
someFunction(
    typeA argA
    typeB argB
)
```
... then the function must be called like this: `ws.some.function(value_a, value_b)`. I.e. with positional arguments like the `leCreatePin` function.

The other case looks like this:

```
anotherFunction(
    ?typeA argA
    ?typeB argB
)
```

In this case you need keyword arguments: `ws.another.function(arg_a=value_a, arg_b=value_b)`.

Note, that these variants could be mixed! The only truth is the SKILL documentation of that function.

JacobEFO, 2020-06-15:
Ah thank you. That was exactly what I needed to know.

I am not sure how common this is within python, but I believe it would be highly useful if the documentation reflected this as well as the 0:0 syntax.

On my end I regard this as resolved, so thank you very much for your attention.

closed 2020-06-15 by JacobEFO