unihd-cag/skillbridge #146: Problem getting started

jfosorio, 2020-07-01:
I am not sure what is PATH-TO-SKILL-IPC-SCRIPT ? Is something that comes with the cadence installation ?

nielsbuwen, 2020-07-01:
Hi,

our documentation seems to be a little bit inconsistent. The `PATH-TO-SKILL-IPC-SCRIPT` is the same thing as the `PATH-TO-IPC-SERVER` and is **not** provided by cadence, but by us. You can obtain this path by typing the command `skillbridge path` into your terminal. 

The command will give you the correct path (it will also provide you with the complete `load` command).

Step-by-step:

1. Have the `skillbridge` installed
2. type `skillbridge path` into your terminal
3. obtain the complete `load(...)` command from the output
4. enter the `load(...)` command into the CIW of your cadence tool
5. enter `pyStartServer` into the CIW

Thank you for your interest in our project and feel free to close the issue as you prefer.

jfosorio, 2020-07-02:
Thanks for the answer it works now.

closed 2020-07-12 by TM90