unihd-cag/skillbridge #147: SKILL++ Support

curtisma, 2020-07-12:
It would be very helpful to be able to run SKILL++ commands also.  I have some SKILL++ libraries that I need access to from Python

Maybe add a way to change the interpreter to skill++, which is mostly backwards compatible with SKILL commands.  This normally done in the CIW using toplevel('ils) command and should be easy to implement by calling that command when the SKILL++ setting is enabled.

There would also need to be support for the -> operator from Python.

SKILL++ allows the user to build software packages in SKILL that enable private functions.  It also has an object system.

nielsbuwen, 2020-07-13:
Hi,

the `~>` Operator is already implemented but it is not documented so far. The usage might be a bit confusing, because we translate SKILL lists directly to python lists.

So you can either write the equivalent in python:

```python
shapes = cv.shapes
print([shape.layer for shape in shapes])
# cv->shapes~>layer
```

Or use [this](https://unihd-cag.github.io/skillbridge/examples/lazy_lists.html) feature:

```python
shapes = cv.lazy.shapes
print(shapes.layer)
```
Concerning the SKILL++ mode: we will look into that. Maybe it already works if you simply call these "ils" function.

curtisma, 2020-07-13:
Ok, sure, Does it also support the SKILL namespace operators "::" and ":::".  These can be used to make packages also.

I'll have to try just calling toplevel('ils) when I start it as you suggested.  My concern is that if it encounters an error it might switch back to skill mode.

nielsbuwen, 2020-07-13:
So the command is literally `toplevel('ils)` in SKILL? In python that would be:

```python
from skillbridge import Workspace, Symbol

# open workspace as usual
ws['toplevel'](Symbol('ils'))
```

TM90, 2020-07-14:
> Ok, sure, Does it also support the SKILL namespace operators "::" and ":::". These can be used to make packages also.

I guess you want to access a function in a specific namespace? 
I did not work with skill++ much before, so how would an example for that look like? 
A test function definition in a namespace would be really useful, to see if we can access it from the skillbridge

curtisma, 2020-07-14:
Yeah, here's an example:

`; First namespace without exporting x
makeNamespace("myNs1")
myNs1:::x =  " World 1"
procedure(myNs1:::hello(in)
	println(strcat("hello1 " in)))

; Second namespace with exporting x
makeNamespace("myNs2")
myNs2:::x = " World 2"
addToExportList('(myNs2:::x))
procedure(myNs2:::hello(in)
	println(strcat("hello2 " in)))

println("")
println("calling each from their namespace")
myNs1:::hello(myNs1:::x)
myNs2:::hello(myNs2::x)
myNs1:::hello(myNs2::x)`

which should return:
`""
"calling each from their namespace"
"hello1  World 1"
"hello2  World 2"
"hello1  World 2"`

nielsbuwen, 2020-07-17:
I am currently testing out a possible implementation of the namespace system. It is available on the `namespaces` branch. Maybe you could comment if it suits your needs:

```python
ws = ...  # open Workspace

# access an existing namespace
ns = ws.namespace('myNs1')

# access the private variable x of the namespace (myNs1:::x)
print(ns['x'])

# change the private variable x of the namespace (myNs1:::x = 123)
ns['x'] = 123

# call a private function of the namespace (myNs1:::x("World"))
ns.hello("World")

# create a new namespace (or use an existing if it already exists)
other_ns = ws.namespace('myNs2')
```

It seems like we don't have to map the `::` (double colon) operator, because the `:::` (triple colon) operator is more powerful.

curtisma, 2020-07-21:
Ok, thanks.  I will let you know when I get a chance to try it out.

closed 2022-08-15 by nielsbuwen

curtisma, 2022-08-15:
@nielsbuwen, did the namespace feature get merged into master?

nielsbuwen, 2022-08-18:
No. Did you get a chance to test the feature?