unihd-cag/skillbridge #149: Conda Package

curtisma, 2020-09-04:
I use Conda as a package manager instead of pip. If I submit a PR request for a conda recipe would you be interested in uploading skillbridge to conda-forge?  Or would you mind if I uploaded it?

nielsbuwen, 2020-09-09:
Hi,

we will take a look at conda-forge. Thank you for the suggestion?

Out of curiosity? What prevents you from installing the `skillbridge` via `pip` in your conda environment?

nielsbuwen, 2022-06-16:
Thank you for the work. I don't believe that i have the capacity (or the experience) at the moment to be a maintainer for the conda recipe. I would be very happy if you could do that.

curtisma, 2022-06-16:
Ok, sounds good.

curtisma, 2022-07-01:
This has been added to [Conda-Forge](https://conda-forge.org/).  
The feedstock can be found at:
https://github.com/conda-forge/skillbridge-feedstock

closed 2022-07-01 by curtisma