unihd-cag/skillbridge #155: port objects are not processed.

psuhan, 2021-01-21:
It looks like skill port objects are not properly processed.

When I open an output port in Python, I code something like this
```python
p = ws['outfile']('./test.txt')
```
But it results in an error
```lisp
port:"./test.txt"
I COULD NOT PARSE "port:\"./test.txt\""
```
Also, it seems like there is no way to access pre-defined port objects such as stdout, stdin, etc., from Python side.
```lisp
p = outfile("./test.txt") 
port:"./test.txt" ;; I opend this port
> 
p = stdout
port:"*stdout*" ;; pre-defined port
>
```
I have no idea exactly how many pre-defined constant objects are there in skill side, but it will be good to have a systematic way to access those objects.

Thanks!!

Pyungsu

TM90, 2021-01-21:
The port issue is a bug and should be fixed. #157

Accessing standard variables or functions could be implemented. We have to find a good approach for this yet. This will be handled in the future in issue #156. 

Why do you want to open files with the skill interface and not directly in python?

psuhan, 2021-01-21:
> The port issue is a bug and should be fixed. #157
> 
> Accessing standard variables or functions could be implemented. We have to find a good approach for this yet. This will be handled in the future in issue #156.
> 
> Why do you want to open files with the skill interface and not directly in python?

I need to call a skill function from Python via skillbrdige. That function is kinda encrypted and requires a port object as its argument. Well, I know I can write a simple skill code to wrap up the function and hide THAT argument inside the wrapper function. In that case, I need to load("wrapper.il") whenever I run >virtuoso &. which is what I am doing now.

Thank you very much for your work. It helps me a lot. I am also learning a lot about Python from your code.

Pyungsu

nielsbuwen, 2021-01-21:
There is a workaround if you want to pass the ´stdout` variable to a SKILL function:

```python
from skillbridge import Workspace, Var

ws = Workspace.open()

ws.prefix.the_function_to_call(Var('stdout'))
```

psuhan, 2021-01-21:
> There is a workaround if you want to pass the `stdout` variable to a SKILL function:
> 
> ```python
> from skillbridge import Workspace, Var
> 
> ws = Workspace.open()
> 
> ws.prefix.the_function_to_call(Var('stdout'))
> ```

I tested it, like
```python
ws['function_accepts_and_returns_port_obj'](Var('stdout'), other args)
```
and I saw that Var('stdout') works beautifully when it is handed as a argument.
Problem is that the function has a port object inside its return value (list), and Python server cannot parse it.

TM90, 2021-01-25:
There is a new [release 1.2.9](https://github.com/unihd-cag/skillbridge/releases/tag/releases%2F1.2.9) 

The global variable handling will be solved in a later issue #156  when we find the time and a good solution.

Can this issue be closed?

TM90, 2021-02-01:
Closed, if this issue becomes relevant again feel free to open it again

closed 2021-02-01 by TM90