unihd-cag/skillbridge #165: direct mode not working

noammeir, 2021-12-26:
hi
I was trying to use skillbridge direct mode by using "ipcBeginProcess("python directBridgeTest.py"), it seems there is no response from the server (I did not activate one...)
python code:

import sys
from skillbridge import Workspace

ws = Workspace.open(direct = True)
cellView = ws.db.open_cell_view_by_type("wk_nmeir", "test", "schematic")
print(cellView)
exit()

output in stderr of the child process:

Traceback (most recent call last):
  File "/nfs/iil/home/nmeir/skill/directBridgeTest.py", line 7, in <module>
    cellView = ws.db.open_cell_view_by_type("wk_nmeir", "test", "schematic")
  File "/nfs/iil/stod/stod903/w.nmeir.100/python_venv/lib/python3.7/site-packages/skillbridge/client/functions.py", line 40, in __call__
    result = self._channel.send(command)
  File "/nfs/iil/stod/stod903/w.nmeir.100/python_venv/lib/python3.7/site-packages/skillbridge/client/channel.py", line 58, in send
    return self.decode_response(input())
EOFError: EOF when reading a line

what am I missing/doing wrong?

TM90, 2021-12-26:
At the moment I am on vacation. I can give you an example in the first week of January.

nielsbuwen, 2021-12-26:
Hi, 

I checked our documentation about the direct mode and couldn't find much. We should really update that.

Calling `ipcBeginProcess` is not enough because then the responses are not handled. There is an undocumented function you can use instead: 

`pyRunScript("yourPythonFile.py")`.

We will add that to the documentation soon.

noammeir, 2021-12-26:
works like a charm Niels, thanks a lot!

closed 2021-12-26 by noammeir