unihd-cag/skillbridge #168: Positional keyword arguments

bou4, 2022-01-04:
How would one implement Ocean's `save` function?

The Ocean reference states the following declaration.
```
save(
[ ?categ s_categ ] s_saveType
[ t_saveName1 ] ... [ t_saveNameN ]
)
=> undefined / nil
```

However, if I would call `ws['save'](Symbol('allv'), categ=Symbol('analog'))`, an error will be thrown at the CIW: `WARNING (OCN-6019): Invalid value ?categ for allv in save. Use ocnHelp('save) for more information.`. Because `?categ` is not the first argument.
Is it an idea to support specifying the position of the keyword arguments to be able to implement this?

Kind regards

Caro Meysmans

bou4, 2022-01-04:
My current workaround is defining a `Keyword` class:
```python
def save(save_type: SaveType = SaveType.allv, names: list[str] = None, category: Category = Category.analog) -> None:
    """Specifies the outputs to be saved and printed during simulation.

    When specifying particular outputs with saveName, you can include as many outputs as you
    want. If you want to turn off the default of save, `allv, use the delete(`save )
    command.

    Arguments:
        save_type:
            Type of outputs to be saved.
            The value can be any member of :class:`SaveType`.
            The default save type is :attr:`SaveType.allv`.
        names:
            List of nets, devices or other objects.
        category:
            Type of simulator to be used.
            The value can be any member of :class:`Category`.
            The default category is :attr:`Category.analog`.
    """
    class Keyword():
        def __init__(self, name: str) -> None:
            self.name = name

        def __repr_skill__(self) -> str:
            return f'?{self.name}'

    if names is None:
        names = []

    return ws['save'](Keyword('categ'), category, save_type, *names)
```

nielsbuwen, 2022-01-04:
Hi Caro,

unfortunately the `save` function is defined in a way that can't be represented directly in python. The problem is, as you found out, the position of the keyword argument.

Luckily we already defined that `Keyword` class. You can import it like this: `from skillbridge import Key`. Your save call then becomes this: `ws['save'](Key('categ'), category, save_type, *names)`

If you can think of a more convenient way, please let us know.

bou4, 2022-01-04:
Hi Niels,

Thanks! I must have missed that. I cannot think of a more convenient way.

Kind regards

Caro

closed 2022-01-04 by bou4