unihd-cag/skillbridge #169: Subtracting waveforms

bou4, 2022-01-04:
Hi all,

I am busy porting part of Ocean to Python to better document the sizing of my circuits in a Jupyter environment.
However, I cannot seem to find a way to subtract two waveforms.
I would like to do something like this:
```python
class Waveform:
  def __sub__(self, other):
     ...
vop = Waveform(ws['getData']('/vop'))
von = Waveform(ws['getData']('/vop'))

vod = vop - von
```
But I don't know how I can execute infix operations, and how I find out which variables are already defined containing these waveforms.
Can you guide me in the right direction?

Kind regards,

Caro Meysmans

bou4, 2022-01-04:
I must have complicated things too much:
```python
class Waveform:
    def __init__(self, waveform) -> None:
        self.waveform = waveform

    def __sub__(self, other: Waveform) -> Waveform:
        return Waveform(ws['plus'](self.waveform, ws['minus'](other.waveform)))

    def pull(self) -> tuple[list[float], list[float]]:
        """Pull the waveform for further processing in the local environment.

        Args:
            waveform: The waveform to pull.

        Returns:
            Tuple containing the x and y values of the waveform.
        """
        x, y = ws['abWaveToList'](self.waveform, transpose=True)

        return x, y
```
Is this the way to go?

nielsbuwen, 2022-01-04:
Hi,

can you share the skill code you are working on? I'm not familiar with these waveforms.

bou4, 2022-01-06:
Hi

I am working on this class `Waveform` which encapsulates a `srrWave` object, as received by Ocean functions like `getData`.
I got it working like this: https://github.com/bou4/ocean/blob/7e045b580bc52601191294f0fc2b1e2423982525/src/ocean/ocean.py#L22, so in fact I do not have a problem anymore.
But I am interested in knowing if this is how you would do it or are there better ways you see fit?

KInd regards

Caro Meysmans

nielsbuwen, 2022-01-08:
Wrapping the Waveforms and implementing the operators on top of that seems like the best option to me.

We could even discuss if we want to add that to this library.

TM90, 2022-01-10:
If each of the operator dunder methods map to a single skill operator it would be fine to add this functionality to the skillbridge.

The operator in question at the moment would be `__sub__` which uses an addition with a minus. 

@bou4 Is it possible to only use minus, i.e. can you write `(minus leftWaveform rightWaveform)` 
instead of `(plus leftWaveform (minus rightWaveform))`?

Best regards

Tobias

bou4, 2022-01-12:
As far as I know, `(minus leftWaveform rightWaveform)` is not possible.

Kind regards

Caro Meysmans

nielsbuwen, 2022-01-12:
That's unfortunate. Then it does not make sense to implement these operators directly in the skillbridge.

I will close the issue. Feel free to reopen it when there is new information

closed 2022-01-12 by nielsbuwen