unihd-cag/skillbridge #176: How to make skillbridge run on tcp port in linux system

kulta-jim, 2022-03-21:
Hi,

I have an idea to develop a website to control a skillbridge service group consisting of multiple servers.

But the problem I have is that I don't know how to connect to the skillbridge service running on another server.

nielsbuwen, 2022-03-21:
Hi,

on Linux we use domain sockets for the communication. To my knowledge there is no way to access that connection from another machine. This was a deliberate decision, because the skillbridge runs user input as code! That is its core feature. But we don't want remote users to connect to the skillbridge. Otherwise we would have to implement authentication and authorization which would be complicated and probably slow things down.

This is a feature we won't implement. But for your project you can fork our repository and make the necessary changes.

The server is created here: https://github.com/unihd-cag/skillbridge/blob/master/skillbridge/server/python_server.py#L64
You should be able to switch the UnixStreamServer with a TCPServer. You also have to use a bind address different from `'localhost'`.

The client is located in https://github.com/unihd-cag/skillbridge/blob/master/skillbridge/server/python_server.py#L64
You can implement it like the windows client which uses TCP.

Let me know if you need more information.

kulta-jim, 2022-03-22:
Okay thank you. Maybe I will seriously consider your suggestion and make some modifications to skillbridge.

closed 2022-03-22 by kulta-jim