unihd-cag/skillbridge #177: Not working when virtuoso is started in silent mode

yegier, 2022-03-24:
Skillbridge is not working when virtuoso is started with -nograph option. I think this behaviour is related to the ipcBeginProcess. It uses stdin to comunicate, and in silent mode it's not activated. Any chance to update it to ipcSkillProcess?

nielsbuwen, 2022-03-28:
Hi,

sorry for the late response. We did some testing and were able to reproduce the issue. At the moment we hesitate to replace `ipcBeginProcess` with `ipcSkillProcess`, because it may introduce some subtle changes.

However the [direct mode](https://unihd-cag.github.io/skillbridge/usage/quickstart.html?highlight=pyrunscript#direct-mode-without-a-server) still works with the `-nograph` option.

In the future we will experiment with `ipcSkillProcess` and hide that behavior behind a flag. Maybe `pyStartServer ?nograph t`.

Greetings,
Niels and Tobias

yegier, 2022-03-28:
Great, thanks for the response, actually it was really fast. I remember it didn't work with direct mode. But I will try again. Let me know when the feature is ready. It's really helpful. 
Thanks!

closed 2022-03-31 by TM90