unihd-cag/skillbridge #178: Quoting Variables

DMKun, 2022-05-31:
how to call this in python? I don't know how to put 'all in python. 

> axlXSectionGet(nil 'all )

Quoting Variables
Quoting is often used with the names of variables and their values. For example, putting a
single quote before the variable a (that is, 'a) prevents a from being evaluated. Instead of
returning the value of a, the name of the variable a is returned when 'a is evaluated.

nielsbuwen, 2022-05-31:
Hi,

looks like we didn't document this. Thank you for pointing that out.

In this simple case you can use a `Symbol`.

```python
from skillbridge import Workspace, Symbol

ws = Workspace.open()
ws.axl.x_section_get(None, Symbol('all'))
```

Let me know if that works for you. We will add that as an example to the documentation.

nielsbuwen, 2022-05-31:
closes by #179

closed 2022-05-31 by nielsbuwen