unihd-cag/skillbridge #187: Issue with snake_to_camel when trying to modify a CDF parameter

wscherr-cuas, 2022-07-14:
**Problem:**
I tried to modify a CDF parameter on an pcell which had the form "xxx_yyy" (this is part of a PDK I can't modify).
When using "ws.inst.xxx_yyy = val" in Python, it gets translated to "inst->xxxYyy = val".
So I was actually not able to change the existing parameter but it added a new (useless) one.

**Workaround**:
I added an additional check in the "snake_to_camel" function in client/translator.py,
when I am using "ws.inst.xxx_yyy_ = val" (underscore at the end) it translates to "inst->xxx_yyy = val":
```
def snake_to_camel(snake: str) -> str:
    if snake.endswith('_') and len(snake)>2:
        return snake[:-2]
    if snake.startswith('_') or '_' not in snake:
        return snake
    return sub(r'_[a-zA-Z]', _upper_without_first, snake)
```
Not really pretty, as it may cause other implications - although, one can use "ws.inst.xxx_yyy__" 
(two underscores at the end) to again get "inst->xxx_yyy_" if needed.

**Proposed solution**:
Ideally, it might be useful to have an option to disable such translations completely.
Especially when one is not using this form but always uses the "camel style" in the code.

nielsbuwen, 2022-07-14:
Hi,

thank you for the issue. We have a similar case where we don't want the snake case translation to happen.

To make it consistent with the other feature we could offer this syntax:

```python
ws = Workspace.open()
inst = ws.get.the_cdf_thingy()
inst['xxx_yyy'] = "value"
print(inst['xxx_yyy'])
```

This will be translated as `inst->xxx_yyy`.

Would that solve your issue? We will discuss whether we want to have that feature, soon.

wscherr-cuas, 2022-07-15:
Yes, item assignment is a good idea! 
This also allows handling element parameters in a more flexible way, I'd say...

Thanks!

wscherr-cuas, 2022-07-15:
Yes, it works! Thanks a lot for your fast response.

closed 2022-07-15 by wscherr-cuas

TM90, 2022-07-15:
Is merged and released with: [1.2.18](https://github.com/unihd-cag/skillbridge/releases/tag/releases%2F1.2.18)