unihd-cag/skillbridge #190: SKILL Table Representation

curtisma, 2022-08-02:
Does skillbridge support reading SKILL tables?  
If not, I think it would make sense to represent them as a dictionary in Python.

I'm trying to access a SKILL table from python, but my python code hangs when I try to read it in:

```python
skill_import = ws.__.x
```

where x has been defined as a global variable in Virtuoso as follows:

```scheme
x = makeTable("test table")
table:test table
x["key"] = "value"
     "value"
x->??
     ("key" "value")
```

nielsbuwen, 2022-08-03:
Currently this does not work. But we will add support for tables and vectors

curtisma, 2022-08-03:
Hopefully it can support using symbols as the keys.  The keys and values can, in general, be any type.

```scheme
x = makeTable("test table")
     table:test table
x['key] = "value"
     "value"
x->??
     (key "value")
```

So then in python:

```python
x = ws.__.x
val = ws.__.x[Symbol("key")]
```

curtisma, 2022-08-03:
Also, thanks so much for looking at it right away!  👍🏻 

It is an important feature to enable the use of the [Virtue SKILL++ framework](https://github.com/cascode-labs/virtue) package Import system.  

I was also hoping to implement a way to access functions from a package using the Virtue package import system.  It would work similar to the existing prefix methodology but take the package from the Virtue Import table and the DPL / table-based packages it contains.

I'm working on updating the docs for it to be more descriptive about how it works and can post here when their ready if there's interest.

nielsbuwen, 2022-08-04:
The current implementation does allow any object that can be send to SKILL as a key. You can test the [here](https://github.com/unihd-cag/skillbridge/pull/191).

I would be interested in any work that uses the `skillbridge`. Feel free to mention me.

curtisma, 2022-08-05:
Thanks @nielsbuwen!

I'm working on testing it out and will be sure to provide feedback.

curtisma, 2022-08-09:
Hey @nielsbuwen, the skill server doesn't seem to start with this branch.

```lisp
Skill Bridge server started successfully, id = 'ids_dev-mayberc'.
PYTHON:   File "/prj/ids_dev/work_libs/mayberc/skill/skillbridge/skillbridge/server/python_server.py", line 21
    def send_to_skill(data: str) -> None:
                          ^
SyntaxError: invalid syntax

server was stopped with code 1
```

nielsbuwen, 2022-08-09:
This looks like the python version you are using is too old. The minimum supported version for skillbridge is 3.6. But that hasn't changed with this feature.

curtisma, 2022-08-09:
Yeah, just forgot to activate my environment before starting Virtuoso.

curtisma, 2022-08-09:
Now the table access seems to be working.

In SKILL I can access the "properties" of a DPL like this:

```list
pkg_dpl = list(nil
                 'SomeFunction SomeFunction
                 'OtherFunction OtherFunction)
pkg_dpl->SomeFunction()

pkg_table = makeTable("My table package")
pkg_table['SomeFunction] = SomeFunction
pkg_table['OtherFunction] = OtherFunction

pkg_table->SomeFunction()
````

Is there an easy way to do the "->" property access operator in skillbridge for both cases?

curtisma, 2022-08-09:
ok, it looks like you can use the "get" method.

However, it is rather verbose compared to the syntax directly in SKILL:

```lisp
ws["get"](ws.__.pkg_table, Symbol("SomeFunction"))
```

vs

```lisp
pkg_table ->SomeFunction
```

curtisma, 2022-08-09:
Is there some operator that could be overloaded in Python for the "->" (get property) skill operator?
Maybe just use the "." operator when we have a remote table or DPL?

So we could have access the function as follows:

```lisp
 ws.__.pkg_table.SomeFunction()
```

curtisma, 2022-08-09:
Also, do you think we could somehow have a unary operator such as "~" be shorthand for the Symbol function?  I've found it's used quite a lot.

curtisma, 2022-08-09:
This should also work for tables at least, and it's a bit better than using get:

```lisp
ws.__.pkgTable[Symbol("someFunction")]()
```

curtisma, 2022-08-09:
One thing I noticed is that when I create a DPL and then read it in Python it converts it to a dictionary but the symbol keys are converted to strings instead of staying as Symbols in Python.  Was that intentional?  I was expecting it to stay as a symbol.

In SKILL:
```lisp
z = list(nil 
            'test 1 
            'test2 "yo")
```

In Python:
```python
ws.__.z
=>
{'test2':'yo', 'test':1}
```

nielsbuwen, 2022-08-10:
> Also, do you think we could somehow have a unary operator such as "~" be shorthand for the Symbol function? I've found it's used quite a lot.

We cannot define or overload operators for builtin types. So `~"xyz"` is invalid in python.

If you need a shortcut, you have two options:

```python
# alias the Symbol class
S = Symbol
print(S("xyz"))

# a dummy class that creates symbols on __getattr__

class SymbolFactory:
    def __getattr__(self, attr):
        return Symbol(attr)

S = SymbolFactory()
print(S.xyz)
```

> One thing I noticed is that when I create a DPL and then read it in Python it converts it to a dictionary but the symbol keys are converted to strings instead of staying as Symbols in Python. Was that intentional?

Yes, that was intentional, because only a list with symbol "keys" is DPL. The translation is bidrectional however:

```python
# SKILL: x = list(nil 'a 1)
print(ws.__.x)  # {'a': 1}

ws['printf']("%L\”", {'a': 1, 'b': 2)
# SKILL (nil 'a 1 'b 2)
```

> `ws["get"](ws.__.pkg_table, Symbol("SomeFunction"))`

Just use `ws.__pkg_table[Symbol("SomeFunction")]`

What we could discuss is adding the `__getattr__` operator for tables:

```python
ws.__.my_table.xyz
# equivalent to ws.__.my_table[Symbol('xyz')]
```

curtisma, 2022-08-10:
Great, thanks for the help @nielsbuwen!

curtisma, 2022-08-10:
> What we could discuss is adding the `__getattr__` operator for tables:
> 
> ```python
> ws.__.my_table.xyz
> # equivalent to ws.__.my_table[Symbol('xyz')]
> ```

Yeah, I think that would be a good feature.

curtisma, 2022-08-11:
Is the associated PR ready to merge?

PR #191 (2022-08-12): Add support for tables and vectors
Closes #190 

- [x] Implementation
- [x] Tests
- [x] Documentation

Example:

```python
ws = Workspace.open()

table = ws.make_table('MyTable', default=None)
# or table = ws.__.my_existing_table

print(table['x'])
table['x'] = 1
del table['x']
```

The `RemoteTable` basically behaves like a python `dict`
closed 2022-08-12 by TM90