unihd-cag/skillbridge #192: ValueError: invalid literal for int() with base 10:

DMKun, 2022-08-31:
if return large data. skillbridge will throw exception.
I'm use cadence 17.2, Allegro Package Designer L.
input code:
```
comps = ws['axlDBGetDesign']().components
dir(comps[0])
```
output:
```
Traceback (most recent call last):
  File "/home/.pycharm_helpers/pydev/_pydevd_bundle/pydevd_exec2.py", line 3, in Exec
    exec(exp, global_vars, local_vars)
  File "<input>", line 1, in <module>
  File "/local/usr/lib/python3.9/site-packages/skillbridge/client/objects.py", line 101, in __getattr__
    result = self._send(self._translate.encode_getattr(self._variable, key))
  File "/local/usr/lib/python3.9/site-packages/skillbridge/client/objects.py", line 77, in _send
    return self._channel.send(command).strip()
  File "/local/usr/lib/python3.9/site-packages/skillbridge/client/channel.py", line 157, in send
    return self._receive_only()
  File "/local/usr/lib/python3.9/site-packages/skillbridge/client/channel.py", line 150, in _receive_only
    received_length = int(received_length_raw)
ValueError: invalid literal for int() with base 10: b'success No'
```

nielsbuwen, 2022-08-31:
Hi, thank you for the bug report. We will investigate it.

DMKun, 2022-09-04:
[image]
I'm using pycharm. when I click right comps->[1]. It's show the error message
the mcm is` SPB_17.4\share\pcb\toolbox\getting_started\panelization\mcm_context\databases\sample.mcm`

nielsbuwen, 2022-09-04:
Thank you for the additional information.

We mainly use the skillbridge for Virtuoso so it will take us a  while to set up an Allegro environment.

Sorry for the delay.

Could you paste the whole error message here, please? The screenshot only shows a part of it.

DMKun, 2022-09-04:
```
D:\source\skillbridge_example\venv\Scripts\python.exe "C:/Program Files/JetBrains/PyCharm 2022.2.1/plugins/python/helpers/pydev/pydevconsole.py" --mode=client --host=127.0.0.1 --port=58293 
import sys; print('Python %s on %s' % (sys.version, sys.platform))
sys.path.extend(['D:\\source\\skillbridge_example', 'D:\\source\\skillbridge_example', 'D:/source/skillbridge_example'])
PyDev console: starting.
Python 3.10.6 (tags/v3.10.6:9c7b4bd, Aug  1 2022, 21:53:49) [MSC v.1932 64 bit (AMD64)] on win32
from skillbridge import Workspace
g_ws = Workspace.open(workspace_id=7777)
comps = g_ws['axlDBGetDesign']().components
ic = comps[1]
p0 = ic.parent_groups[0]
attempting to reconnectattempting to reconnect
attempting to reconnect
p1 = ic.parent_groups[1]
Traceback (most recent call last):
  File "C:\Program Files\Python310\lib\code.py", line 90, in runcode
    exec(code, self.locals)
  File "<input>", line 1, in <module>
  File "D:\source\skillbridge_example\venv\lib\site-packages\skillbridge\client\objects.py", line 97, in __getattr__
    result = self._send(self._translator.encode_getattr(self._variable, key))
  File "D:\source\skillbridge_example\venv\lib\site-packages\skillbridge\client\objects.py", line 76, in _send
    return self._channel.send(command).strip()
  File "D:\source\skillbridge_example\venv\lib\site-packages\skillbridge\client\channel.py", line 157, in send
    return self._receive_only()
  File "D:\source\skillbridge_example\venv\lib\site-packages\skillbridge\client\channel.py", line 150, in _receive_only
    received_length = int(received_length_raw)
ValueError: invalid literal for int() with base 10: b'y_dbid_000'

```

DMKun, 2022-09-04:
this is show in allegro
```
ERROR
print(1)
command "        35__py_dbid_000002D33996722\r" resulted in error ("lineread/read" 1 t nil ("*Error* lineread/read: syntax error encountered in input at line 1 column 11 in file *stdin*"))
1
nil
print(1)
1
nil
```

DMKun, 2022-09-04:
[skillbridge_server.log](https://github.com/unihd-cag/skillbridge/files/9485050/skillbridge_server.log)
[skillbridge_skill.log](https://github.com/unihd-cag/skillbridge/files/9485051/skillbridge_skill.log)

nielsbuwen, 2022-09-08:
Hi, i got the chance to debug the windows-channel implementation. Unfortunately i could not reproduce your problem. How big is the data you are sending?

Did you try to cancel a running skillbridge session with `Ctrl+C`? That can confuse the synchronization between client and server. The error you are getting looks like that.

nielsbuwen, 2022-09-08:
Good hint, i will try it out in PyCharm and its debugger when i get a chance

DMKun, 2022-09-08:
[image]

DMKun, 2022-09-08:
[image]
Maybe, It's not a problem.

DMKun, 2022-10-12:
[image]
pycharm set "variables loading policy"->"On demand" can fix this problem.

TM90, 2022-10-16:
> [image] pycharm set "variables loading policy"->"On demand" can fix this problem.

Thats a great tip. Thank you!

Can this issue be closed, are there any open problems regarding this issue?

closed 2022-10-27 by TM90