unihd-cag/skillbridge #193: Global Variables filter

DMKun, 2022-09-20:
ws = Workspace.open()

my_globals = ws.globals('my_globals')
my_globals.shapes << ws.ge.get_sel_set.var()

# get the boxes
my_globals.boxes << my_globals.shapes.map(loop_var.b_box)

# filter rectangles
my_globals.shapes << my_globals.shapes.filter(loop_var.obj_type == 'rect')
my_globals.shapes << my_globals.shapes.filter(loop_var.isRect==True)
my_globals.shapes << my_globals.shapes.filter(loop_var.isHole==True)

How to merge multiple filter.

nielsbuwen, 2022-09-22:
Hi, do you want to filter on multiple conditions? You can use `|` and `&` for that.

```python
my_globals.shapes.filter(loop_var.obj_type == 'rect' | loop_var.obj_type == 'path')
```

That means all shapes that are rects **or** paths.

The **and** (`&`) works identically.

DMKun, 2022-09-24:
It's work, thanks.

closed 2022-09-24 by DMKun