unihd-cag/skillbridge #197: Wire  label name bug

ClarenceChee, 2022-10-20:
When I use skill command(schCreateWire and schCreateWireLabel) in CIW window, the wire name can automatically change to label name I set.
But when I use python codes(sch.create_wire and sch.create_wire_label), the wire name can not change to label name I designated

nielsbuwen, 2022-10-20:
Hi,

this could be a GUI problem. Does the label change when you pan in the schematic view? This causes a redraw.

If that is the problem you can use `ws.hi.redraw()` in python to force the viewport to redraw.

ClarenceChee, 2022-10-21:
> Hi,
> 
> this could be a GUI problem. Does the label change when you pan in the schematic view? This causes a redraw.
> 
> If that is the problem you can use `ws.hi.redraw()` in python to force the viewport to redraw.

Hi nielsbuwen,
Neither ws.hi.redraw() in python or hiRedraw() in skill will fix this problem. But when I manually create a label in schematic and attach it to the wire which python code generates, the wire name will change to the label name.

nielsbuwen, 2022-10-21:
Can you please share the exact SKILL code (that works) and the exact python code (that does not work). Then i can try to reproduce it.

ClarenceChee, 2022-10-21:
SKILL code
```
cv_schematic = dbOpenCellViewByType("test" "label_test" "schematic" "schematic" "w" nil)
net_vdd = schCreateWire(cv_schematic  "draw" "full" list(0:0 1:0) 0.0625 0.0625 0)
schCreateWireLabel(cv_schematic car(net_vdd) (1:0.5) "VDD" "centerRight" "R0" "stick" 0.0625 nil)
```

Python code
```
from skillbirdge import Workspace
ws = Workspace.open()
cv_schematic = ws.db.open_cell_view_by_type("test", "label_test", "schematic", "schematic", "w", "nil")
net_vdd = ws.sch.create_wire(cv_schematic , "draw", "full", [[0,0], [1,0]], 0.0625, 0.0625, 0)
ws.sch.create_wire_label(cv_schematic, net_vdd[0], [1,0.5], "VDD", "centerRight", "R0", "stick", 0.0625, nil)
```

nielsbuwen, 2022-10-21:
What is the `nil` in your python code?

So after you run the python code, `net_vdd[0].net.name` is still "net1" instead of "VDD"?

When i try this, the net is changed to "VDD".

[image]

ClarenceChee, 2022-10-21:
`net_vdd[0].net` is None, so I can't get `net_vdd[0].net.name`.
Is there any possibility that the problem caused by the version of icadv?
What is the version of your virtuoso?

nielsbuwen, 2022-10-21:
Please try your python code again and replace all `"nil"` with `None`.

ClarenceChee, 2022-10-22:
> Please try your python code again and replace all `"nil"` with `None`.

That will solve the problem,  many thanks!

nielsbuwen, 2022-10-22:
You are welcome. Feel free to open another issue when you need help.

closed 2022-10-22 by nielsbuwen