unihd-cag/skillbridge #198: How to Trigger the Callback Function

Bobobozz, 2022-10-25:
Hi
I want to change the "w" and "l" of a instance, and I don't know how to trigger the callback
function.

I  use skill code like:
 (foreach pName list("m" "l" "fw" "fingers")
    param=cdfFindParamByName(cdfgData pName)
        (when param~>callback
            evalstring(param~>callback)
         );when
);foreach

 evalstring(param~>callback) works well in skill, but when I use python code: eval(param.callback),it cames error

nielsbuwen, 2022-10-25:
Hi,

i assume that there is SKILL code in `param.callback`. In that case you have to call the SKILL `evalstring` function.

```python
from skillbridge import Workspace

ws = Workspace.open()
ws['evalstring'](...)
```

Bobobozz, 2022-10-25:
> Hi,
> 
> i assume that there is SKILL code in `param.callback`. In that case you have to call the SKILL `evalstring` function.
> 
> ```python
> from skillbridge import Workspace
> 
> ws = Workspace.open()
> ws['evalstring'](...)
> ```

You are right! Thanks!

closed 2022-10-25 by Bobobozz

leochand101, 2023-07-06:
Can you provide a simple example of changing width or number of fingers and run a callback after that, when I run the commands above I dont see the ad/as source or drain area updating.

Bobobozz, 2023-07-06:
@leochand101  
You can use like this after setting parameters.
[image]

leochand101, 2023-07-06:
What is self here (i.e which object instance) 
I access the individual parameters for an instances using <inst_obj>.prop which return the prop object, I guess I need to know what self is in this context, My update of prop.value shows up in the schematic but there is a problem of saving it without updating the derived parameters using a callback, hence the original question

leochand101, 2023-07-07:
UPDATE
Looks like the technology does not allow params of instances to be updated only of pcells themselves, trying it on a different technology now, still havent got he callbacks updating anything.