unihd-cag/skillbridge #199: How to Identify a Global Variable in Skill

Bobobozz, 2022-10-26:
cdfgData is  a global variable in skill, and I want to change the value of cdfgData using python. How can I call the Global Variable in Skill by using python?
I have tried that use the variable names directly in Python,but it seems doesn't work.
skill code:
`cdfgData = cdfGetInstCDF(inst)`
python code:
`cdfgData = ws.cdf.get_Inst_CDF(inst)`

nielsbuwen, 2022-10-26:
Hi,

if you really need global variables, you can use a skill function to set them:

```python
from skillbridge import Workspace, Symbol

ws = Workspace.open()

cdfg_data = ws.cdf.get_inst_CDF.var(...)
ws['set'](Symbol('cdfgData'), cdfg_data)
```

When you don't need the variable in python, it is possible to eliminate one round-trip:

```python
ws['set'](Symbol('cdfgData'), ws.cdf.get_inst_CDF.var(...))  # notice the .var
```

Bobobozz, 2022-10-27:
Hi nielsbuwen,
Thank you for your reply! It works well!

closed 2022-10-27 by Bobobozz