unihd-cag/skillbridge #200: Two users can not use skillbridge at the same time on on machine

ClarenceChee, 2022-10-31:
Once start pyStartServer， a file /tmp/skill-server-default.sock  will be created, if another user want to use skill bridge on this machine, there will be an error:
PermissionError:  [Errno 1] Operation not permitted: '/tmp/skill-server-default.sock'.
Maybe the file should be created as /tmp/$user/skill-server-default.sock

TM90, 2022-10-31:
Hey,

Using one virtuoso instance with multiple users is not possible since we pipe commands to the Virtuoso CIW. And this will get mixed up if there are multiple clients.

What is possible is to start multiple virtuoso instances with different servers where different clients can connect to.

For this use case you can start a virtuoso server with an id and connect to the server with this id in the client.

https://unihd-cag.github.io/skillbridge/examples/multiple-instances.html

Hope that helps.

Best regards

ClarenceChee, 2022-10-31:
> Using one virtuoso instance with multiple users is not possible since we pipe commands to the Virtuoso CIW. And this will get mixed up if there are multiple clients.

Thanks for your reply.  Yes we did start a virtuoso server with an id now as a alternative solution.  
I am not mean to "Using one virtuoso instance with multiple users",  but multiple users work at one machine with their own different virtuoso instances.  Since every user will create a file /tmp/skill-server-default.sock, there will be a conflict, I am wondering if the file can be created as /tmp/$user/skill-server-default.sock to avoid the conflict between different users?

TM90, 2022-10-31:
> > Using one virtuoso instance with multiple users is not possible since we pipe commands to the Virtuoso CIW. And this will get mixed up if there are multiple clients.
> 
> Thanks for your reply. Yes we did start a virtuoso server with an id now as an alternative solution. I am not mean to "Using one virtuoso instance with multiple users", but multiple users work at one machine with their own different virtuoso instances. Since every user will create a file /tmp/skill-server-default.sock, there will be a conflict, I am wondering if the file can be created as /tmp/$user/skill-server-default.sock to avoid the conflict between different users?

Ok now i understand the problem.

We will look into in the course of the week.

Best regards

TM90, 2022-11-08:
For your use-case we recommend using ids. We will not automatically create ids based on users, since on Windows machines the communication and ID mechanism is quite different and we cannot do the same there.

If you want to simplify usage, we recommend using a slim skill wrapper function which uses the current user as id to start the server.

Best regards

Tobias

closed 2022-11-09 by TM90

PiyushSaini07, 2023-08-08:
Just curious with the  `/tmp/skill-server-default.sock` file - this is created once I start the python server. Is there a way to delete it automatically?
Right now I'm deleting the file manually.

> If you want to simplify usage, we recommend using a slim skill wrapper function which uses the current user as id to start the server.

I'm doing this for multiple users.