unihd-cag/skillbridge #202: need Keep Underline

DMKun, 2022-12-02:
Keep Underline
python
``` python
apd = ws.globals(f'rule')
apd.ccc << apd.items.map(loop_var['prop']['ABC_DEF'])
```
skill
```skill
ruleCcc = mapcar(lambda((i) i->prop->ABCDEF) ruleItems ) nil
```
expect
`ruleCcc = mapcar(lambda((i) i->prop->ABC_DEF) ruleItems ) nil`

nielsbuwen, 2022-12-02:
I am not quite sure, why the `__geitem__` operator also performs snake-to-camel conversion.

This could count as a bug. We will discuss that.

DMKun, 2022-12-05:
```
class Var:
    def __getitem__(self, item: Union[str, int]) -> 'Var':
        if isinstance(item, str):
            return Var(f'{self.name}->{snake_to_camel(item)}')
        return Var(f'nth({item} {self.name})')
```

Maybe removing snake_to_camel is the best choice。
sub(r'_[a-zA-Z]', _upper_without_first, snake) cannot compatible with ABC_DEF and axlPolyFromDB

```
        a = ws['axlPolyFromDB'].var()
        b = ws.axl.poly_from_DB.var()
        c = loop_var['ABC_DEF ']
```
expect
```
axlPolyFromDB( )
axlPolyFromDB( )
i->ABC_DEF 
```

PR #209 (2022-12-16): 202 snake
Closes #202
closed 2022-12-16 by TM90