unihd-cag/skillbridge #205: slotValue error with selectResult

appwagner, 2022-12-15:
When I issue ws['selectResult'](Symbol('tran)) I get the following error message: 

command "__py_stdobj_0x335f4d10->objType" resulted in error ("slotValue" 0 t nil ("*Error* slotValue: no such slot - objType in class famParamAnalysis. Valid slot names: (dataFile name wave obj type sweepValue _pDistoHarmonicListCache _pDistoHarmonicFreqListCache _pacHarmonicFreqListCache sweepName sweepValues sweepTable analysisSignals)"))

but when I issue the equivalent selectResult('tran) in the SKILL terminal I get no error. As far as I can tell ws['selectResult'](Symbol('tran)) still works in that I have access to the underlying data in the result but error message is making some other debugging I'm trying to do difficult.

nielsbuwen, 2022-12-15:
Hi,

it looks like the error only occurs when you print the result. Can you confirm that?

nielsbuwen, 2022-12-16:
And could you please check that you are using the most recent version of skillbridge? We fixed a similar issue in the past.

appwagner, 2022-12-16:
I get the error message when I run ws['selectResult'](Symbol('tran)) in a Jupyter Notebook cell. I can try setting it equal to something and see if that suppress the error. Does selecResult return a bool? Also the error was generated with Skillbridge 1.3.1, I just upgraded to 1.4 .

appwagner, 2022-12-16:
Ok, upgrading to Skillbridge 1.4 seems to have fixed that error, thanks for the help.

nielsbuwen, 2022-12-16:
No problem. Feel free to open new issues when something comes up.

closed 2022-12-16 by nielsbuwen