unihd-cag/skillbridge #210: [SUPPORT] General understanding of the capabilities

plifzig, 2023-02-04:
I found 'skillbridge' with a google search 'skill python cadence' and I am thinking it could solve my issue.

I've created a tkinter dashboard GUI for keeping track of layout verification results.  The last piece of my puzzle is to figure out how to launch Calibre RVE from a tkinter button click and being able to highlight RVE errors in the current Virtuoso Layout session.

Can this code help me implement this?

nielsbuwen, 2023-02-04:
Hi,

thank you for contacting us.

Do you have a working SKILL code example that "launches Calibre RVE"? We can help you translate that into Python using the skillbridge.

However if you don't know how to do it in SKILL, i won't be able to help you, because my knowledge of the cadence tools is limited.

So basically: if you can do it with SKILL code then you can also do it with skillbridge.

TM90, 2023-02-04:
Hi,

So as @nielsbuwen mentioned the skillbridge is a translater between python and skill.

Everything within virtuoso which has a skill interface can be controlled.

I am quite sure that there is no Calibre Skill Interface since Calibre is a Siemens tool and Virtuoso/Skill Cadence.

Calibre is a standalone tool and I would guess that it is possible to run it from the command line. At least I found some hints about this [here](https://www.edaboard.com/threads/running-calibre-drc-and-lvs-from-command-line.133452/). I never used Calibre so I have to guess here a bit.

So a better way to trigger it from python would be building a calibre command and executing and controlling it with [subprocesses](https://docs.python.org/3/library/subprocess.html)

Hope this helps.

Best regards

Tobias

plifzig, 2023-02-04:
Thanks for the replies, I appreciate it.

In my CAD environment there are some Calibre env variables set which provide access to the following SKILL function (I use this in some SKILL scripts, myself):

mgc_calibre_start_rve("/the/path/to/the/file.results")

I will look into those links and see if they make sense to me -- I am a novice hack of a programmer with no formal training, so some trivial things can go over my head :)

TM90, 2023-02-04:
> mgc_calibre_start_rve("/the/path/to/the/file.results")
> 
> I will look into those links and see if they make sense to me -- I am a novice hack of a programmer with no formal training, so some trivial things can go over my head :)

It is possible to call custom skill methods from skillbridge with the so called direct [method](https://unihd-cag.github.io/skillbridge/examples/custom_functions.html#direct-method)

so in your case:

``` python
from skillbrige import Workspace
ws = Workspace()
ws['mgc_calibre_start_rve']("/the/path/to/the/file.results")
```

But still while it is possible this solution has several disadvantages in comparision with calling calibre in batch/cli mode from a subprocess:

- you are using virtuoso and calibre and with it for both a license
- With the interaction from python -> virtuoso -> calibre it will be more difficult to read the calibre tool output...

TM90, 2023-02-09:
We would close the issue. If you have any further questions feel free to reopen it.

closed 2023-02-09 by TM90