unihd-cag/skillbridge #211: [FEATURE] Mypy now supports recursive types

nielsbuwen, 2023-03-07:
**Is your feature request related to a problem? Please describe.**

Skill functions are typed as accepting `Any` and returning `Any`. This is annoying, because it is too general

**Describe the solution you'd like**

Define the possible return type and argument types as a recursive type alias.

**Describe alternatives you've considered**

none

DMKun, 2023-03-13:
I have considered defining the data type of the return value in Skill and then corresponding the return value with the data type in workspace.pyi. In this way, I can use the auto-completion function to write code and also have some type checking function.
such as axlDBGetDesign return o_design.
我曾想过定义Skill里的返回值的数据类型，然后在workspace.pyi将返回值与数据类型对应上。这样我可以使用自动补全功能编写代码，也有一定的类型校验功能。

现在我的skill代码已经几乎写完了，我现在更关心如何实现skill回调python，或者如何提升性能(多重循环不使用skill代码来提升性能)。
Now my skill code is almost finished, and I am more concerned about how to implement skill callback python, or how to improve performance (not using skill code to improve performance in multiple loops).

TM90, 2023-03-16:
> I have considered defining the data type of the return value in Skill and then corresponding the return value with the data type in workspace.pyi. In this way, I can use the auto-completion function [...]

It sound like you are talking about the dynamic auto-completion functionallity? This isssue is more about better and a more realistic static typing with mypy.

> 现在我的skill代码已经几乎写完了，我现在更关心如何实现skill回调python，或者如何提升性能(多重循环不使用skill代码来提升性能)。 Now my skill code is almost finished, and I am more concerned about how to implement skill callback python, or how to improve performance (not using skill code to improve performance in multiple loops).

If you want to improve performance without using skill code this might be interesting to you:

[lazy_lists](https://unihd-cag.github.io/skillbridge/examples/lazy_lists.html)

nielsbuwen, 2023-07-27:
Let's defer this until version 2.0

Changing all the type hints will be a backwards incompatible change

closed 2023-07-27 by nielsbuwen