unihd-cag/skillbridge #215: [SUPPORT] Server wont return a response

leochand101, 2023-04-05:
**I had the server working, after I quit and reopened virtuoso, I am running into this error**

  ```
File "/home/<>/.local/lib/python3.9/site-packages/skillbridge/client/channel.py", line 40, in decode_response
    status, response = response.split(' ', maxsplit=1)
ValueError: not enough values to unpack (expected 2, got 1)
```

I have a local install of skillbridge which I was able to interact through pyCharm just fine. 
Now as I source the script and start the server I get the above error
I started the new server as : 
 pyStartServer ?id "$USER" ?logLevel "INFO" ?python "LD_LIBRARY_PATH= python3.6"
 I dont have /usr/bin/python so I used python3.6 instead which I had, not sure if that is the cause.

When I run pyShowLog I see below (clearly an issue)
```
unboundunboundunboundunboundunboundunboundunboundunboundunboundunboundunboundunboundunboundunboundunboundunboundunboundunboundunboundunbound
t
```
Any help would be greatly appreciated!!

nielsbuwen, 2023-04-05:
The error you posted occurs when you reopen Virtuoso **without** starting the python server again.

Did you run `pyStartServer ...` again after the restart?

leochand101, 2023-04-05:
Ok, I debugged it and found that the **id that I was providing** was clearly causing issues I just provided pyStartServer ?python "<path to python>" and this worked.

closed 2023-04-05 by leochand101