unihd-cag/skillbridge #216: [BUG] RuntimeError on ws.db,save

leochand101, 2023-04-06:
**Describe the bug**
`RuntimeError: ("putprop" 0 t nil ("*Error* putprop: first arg must be either symbol, list, defstruct or user type" nil))` while running ws.db.save

**To Reproduce**
ws.db.save(cell_dict[topcell]) I am passing a cellview_pointer to it

**Expected behavior**
The command clearly works as the schematic is saved but then the return value has some issues, I believe its expecting

**Error Stacktrace**
Response is ("putprop" 0 t nil ("*Error* putprop: first arg must be either symbol, list, defstruct or user type" nil))
 Status is failure
Traceback (most recent call last):
  File "/home/<>/scripts/Query_cadence.py", line 287, in <module>
    set_width("w", "500n", cell_comp_dict, topcell)
  File "/home/<>/scripts/Query_cadence.py", line 236, in set_width
    ws.db.save(cell_dict[topcell], cell_dict[topcell].lib_name, cell_dict[topcell].cell_name, cell_dict[topcell].view_name)
  File "/home/<>/.local/lib/python3.9/site-packages/skillbridge/client/functions.py", line 41, in __call__
    result = self._channel.send(command)
  File "/home/<>/.local/lib/python3.9/site-packages/skillbridge/client/channel.py", line 158, in send
    return self._receive_only()
  File "/home/<>/.local/lib/python3.9/site-packages/skillbridge/client/channel.py", line 154, in _receive_only
    return self.decode_response(response)
  File "/home/<>/.local/lib/python3.9/site-packages/skillbridge/client/channel.py", line 48, in decode_response
    raise RuntimeError(response)
RuntimeError: ("putprop" 0 t nil ("*Error* putprop: first arg must be either symbol, list, defstruct or user type" nil))

**Additional context**
Save happenend succesfully but the return value evaluates to failiure.
Not sure if I am using it wrong or not because the cell was saved.

nielsbuwen, 2023-04-06:
Unfortunately i cannot reproduce your error.

This code works for me:

```python
from skillbridge import Workspace

ws = Workspace.open()

c = ws.ge.get_edit_cell_view()
assert c is not None

print(ws.db.save(c, c.lib_name, c.cell_name, c.view_name))  # True
```

Are you sure that `cell_dict[topcell]` is a valid `<remove cellView@...>`? Can you show the contents of that `cell_dict` and also the values  `lib_name`, `cell_name` and `view_name`.

Another idea: Did you call the function `putprop` anywhere? Maybe you ran it directly in the CIW.

leochand101, 2023-04-06:
Ok I see what you said here, I did change some cell masters using replace_any_masters and hence the pointer to the cell is no longer the same I needed to open the modified cell again using ws.db.open_cell_view which returns the updated pointer and that saves fine!!!
Thanks @nielsbuwen 
**Solution for anyone ever encountering this error, reopen the cell in question again after modification to save it successfully :**

leochand101, 2023-04-06:
Can you update the label from BUG to SUPPORT @nielsbuwen ?

closed 2023-04-06 by leochand101