unihd-cag/skillbridge #217: [SUPPORT] double looping performance

DMKun, 2023-04-17:
我有一个函数，获取IC和IC边框大小信息。这个函数现在涉及到了双重循环。我应该如何提升函数的执行性能。
已知的方式是将get_die_rect函数变为skill代码，然后通过ws['getDieRect']获取制定对象。

I have a function that return information of the IC and IC border. This function now involves double looping. How can I improve the performance of the function?

One known way is to convert the 'get_die_rect' function into Skill code and then access the specified object through 'ws['getDieRect']'.

Is there a way to improve the performance solely through Python code?
```
def get_ic_and_bBox(ws):
    apd = ws.globals(f'skillBridge')

    design_var = ws.axl.DB_get_design.var()
    apd.comps << design_var.components

    apd.dies << apd.comps.filter(
        (loop_var['class'] == 'IC')
        & (loop_var['symbol'] != None)
    )

    #
    def get_die_rect(items):
        for item in items:
            if item['layer'] == "COMPONENT GEOMETRY/ASSEMBLY_TOP":
                return item

    apd.die_infos << apd.dies.map({
        'name': loop_var['name'],
        'children': loop_var['symbol']['children'],
    })

    die_infos = apd.die_infos()

    for die_info in die_infos:
        die_info['bBox'] = get_die_rect(die_info['children'])['bBox']

    return apd.die_infos
```

TM90, 2023-04-17:
This might help you:

https://unihd-cag.github.io/skillbridge/examples/lazy_lists.html#handling-large-lists

nielsbuwen, 2023-04-17:
This is tricky. Your inner loop is some kind of `find` action. As you probably already discovered, we only mapped the `map`, `filter`  and `for_each` actions. I will investigate if such a function exists in SKILL.

nielsbuwen, 2023-04-17:
There is an `exists` function. Your inner loop would look like that: `car(exists(item items item->layer == "..."))`. The `car` part is a bit annoying, but you should be able to write that python loop completely lazy.

```python
car = ws['car'].var
exists = ws['exists'].var
item = Var('item')

apd.die_info.map(car(exists(item, loop_var, item.layer == 'COMPONENT ...')))
```

Unfortunately, I cannot try this out at the moment. I hope this helps.

DMKun, 2023-04-17:
> This might help you:
> 
> https://unihd-cag.github.io/skillbridge/examples/lazy_lists.html#handling-large-lists

globals var is similar to lazy_lists. So I don't think it's useful for nested loops.

DMKun, 2023-04-17:
> There is an `exists` function. Your inner loop would look like that: `car(exists(item items item->layer == "..."))`. The `car` part is a bit annoying, but you should be able to write that python loop completely lazy.
> 
> ```python
> car = ws['car'].var
> exists = ws['exists'].var
> item = Var('item')
> 
> #                            inner loop var    outer loop var
> #                                         v        v
> apd.boxes = apd.die_info.map(car(exists(item, loop_var.children, item.layer == 'COMPONENT ...')))
> print(apd.boxes())  # the list of boxes
> ```
> 
> Unfortunately, I cannot try this out at the moment. I hope this helps.

I got it. But it's not quite what I expected. 
It still requires the developer to have skill development experience. But this should be the best option for now.

closed 2023-04-19 by TM90