unihd-cag/skillbridge #218: [SUPPORT] performance test?

DMKun, 2023-05-08:
How long does it take to execute this code?
I test it in windows assignment() return 0.001s. In linux 0.1s.

```
ws = Workspace.open(workspace_id=workspace)
apd = ws.globals(f'rule')
def assignment():
    start_time = timeit.default_timer()
    n = 10
    for i in range(0, n):
        apd.n << i
        # g_ws["print"](i)
    end_time = timeit.default_timer()
    t1 = end_time - start_time
    return t1/n
```

nielsbuwen, 2023-05-08:
On my linux machine it takes `0.00021` seconds per iteration. Or even `0.0001` seconds when i increase the number of iterations.

I don't have a windows machine where i can test this.

DMKun, 2023-05-09:
It seems that there is something wrong that is causing slow communication on my Linux machine.  
I've try Direct mode, but it still took `0.1` second.

DMKun, 2023-05-09:
Maybe Allegro Package Designer It's too slow.
I'm test in version `17.2 S045`. Version `17.2 S051` has same result.
skillbridge_server.log
```
09.05.2023 11:25:16.884 INFO client  connected
09.05.2023 11:25:16.884 DEBUG got length b'        41'
09.05.2023 11:25:16.884 DEBUG received 41 bytes
09.05.2023 11:25:16.884 DEBUG got data load("/tmp/pycharm_project_459/perf.il" )
09.05.2023 11:25:16.884 DEBUG sent data to skill
09.05.2023 11:25:16.970 DEBUG got response from skill b'success True\n'
09.05.2023 11:25:16.970 DEBUG sent response to client
09.05.2023 11:25:16.970 DEBUG got length b'         9'
09.05.2023 11:25:16.970 DEBUG received 9 bytes
09.05.2023 11:25:16.970 DEBUG got data print(1 )
09.05.2023 11:25:16.970 DEBUG sent data to skill
09.05.2023 11:25:17.070 DEBUG got response from skill b'success None\n'
09.05.2023 11:25:17.070 DEBUG sent response to client
09.05.2023 11:25:17.070 DEBUG got length b'        14'
09.05.2023 11:25:17.070 DEBUG received 14 bytes
09.05.2023 11:25:17.071 DEBUG got data ruleN = -1 nil
09.05.2023 11:25:17.071 DEBUG sent data to skill
09.05.2023 11:25:17.170 DEBUG got response from skill b'success None\n'
09.05.2023 11:25:17.170 DEBUG sent response to client
09.05.2023 11:25:17.170 DEBUG got length b'        13'
09.05.2023 11:25:17.170 DEBUG received 13 bytes
09.05.2023 11:25:17.170 DEBUG got data ruleN = 0 nil
09.05.2023 11:25:17.171 DEBUG sent data to skill
09.05.2023 11:25:17.270 DEBUG got response from skill b'success None\n'
09.05.2023 11:25:17.270 DEBUG sent response to client
09.05.2023 11:25:17.271 DEBUG got length b'        13'
09.05.2023 11:25:17.271 DEBUG received 13 bytes
09.05.2023 11:25:17.271 DEBUG got data ruleN = 1 nil
09.05.2023 11:25:17.271 DEBUG sent data to skill
09.05.2023 11:25:17.370 DEBUG got response from skill b'success None\n'
09.05.2023 11:25:17.371 DEBUG sent response to client
09.05.2023 11:25:17.371 DEBUG got length b'        13'
09.05.2023 11:25:17.371 DEBUG received 13 bytes
09.05.2023 11:25:17.371 DEBUG got data ruleN = 2 nil
09.05.2023 11:25:17.371 DEBUG sent data to skill
09.05.2023 11:25:17.471 DEBUG got response from skill b'success None\n'
09.05.2023 11:25:17.472 DEBUG sent response to client
09.05.2023 11:25:17.472 DEBUG got length b'        13'
09.05.2023 11:25:17.472 DEBUG received 13 bytes
09.05.2023 11:25:17.472 DEBUG got data ruleN = 3 nil
09.05.2023 11:25:17.472 DEBUG sent data to skill
09.05.2023 11:25:17.571 DEBUG got response from skill b'success None\n'
09.05.2023 11:25:17.571 DEBUG sent response to client
09.05.2023 11:25:17.571 DEBUG got length b'        13'
09.05.2023 11:25:17.571 DEBUG received 13 bytes
09.05.2023 11:25:17.571 DEBUG got data ruleN = 4 nil
09.05.2023 11:25:17.571 DEBUG sent data to skill
09.05.2023 11:25:17.672 DEBUG got response from skill b'success None\n'
```

TM90, 2023-05-11:
> Maybe Allegro Package Designer It's too slow. I'm test in version `17.2 S045`. Version `17.2 S051` has same result. skillbridge_server.log
>
> ```
> 09.05.2023 11:25:16.884 INFO client  connected [...]

We mostly use skillbridge with virtuoso and cannot reproduce perfomance issues in Allegro.

In general:

As @nielsbuwen said for loops will always have the communication overhead which becomes a problem for big lists.
A option to reduce this overhead is to use higher order function or try running these loops on skill side. For this we offer some helper methods see [Lazy Lists](https://unihd-cag.github.io/skillbridge/examples/lazy_lists.html)

DMKun, 2023-05-17:
I know  the  communication overhead. I had use Global Variables and lazy to resolving performance issues. However,  the execution time of a skillbridge func spend too long time. This appears to be an issue with my APD software or system.

DMKun, 2023-05-19:
> On my linux machine it takes `0.00021` seconds per iteration. Or even `0.0001` seconds when i increase the number of iterations.
> 
> I don't have a windows machine where i can test this.
> 
> Python for-loops will always be slow, because each iteration has the communication overhead. Better use the higher order functions.

Can you tell me your system version and software version? I want to solve the problem of my machine running a statement slowly.

nielsbuwen, 2023-05-23:
Unfortunately I cannot tell you too much about my work infrastructure. We are not doing anything fancy. Just standard Linux and Virtuoso in a recent version. At its inception i also tried it on my universities computers with similar results.

DMKun, 2023-05-23:
> 

Thanks.  I've tried SPB from 17.2S000 to 17.2S083 and 17.4 in Linux。
17.4 work fine，But all 17.2 version runing slowly。

closed 2023-05-23 by DMKun