unihd-cag/skillbridge #219: [SUPPORT]: order of execution

baptistesion, 2023-06-01:
Hello, first of all I want to thanks you for this amazing tool.
Then here is my issue, I am working on converting some old skill code into python so I have to do it part by part. Hence, I'm working with the direct mode and I'm calling with pyRunScript a python script in the middle of other skill instruction. what seems to happened is that the python scripts is ended after other skill instruction that where after the pyRunScript instruction.
For a simple example with this code skill: 
```
(pyRunScript "test.py")
println("end ")
```
And the test.py code : 
```python
from skillbridge import Workspace
ws = Workspace.open(direct = True)
ws["println"]("toto")
ws.close()
```
The output that I get with that one is :
```
end 
t
toto
script finished with code 0
```
while the expected output would be : 
```
toto
script finished with code 0
end
t
```
Did I miss something in the documentation? or is there something that I can do to "force" the execution of the python script before the end of the skill script?

Best regards,

nielsbuwen, 2023-06-01:
Hi,

> Did I miss something in the documentation?

I believe we didn't document this behavior. `pyRunScript` effectively calls `ipcBeginProcess(...)`. That is a non-blocking operation, so the python script will run in the background.

You can however explicitly wait for the process to finish by using `ipcWait`. So in your case:

```lisp
(ipcWait (pyRunScript "test.py"))
println("end ")
```

should do the trick.

@TM90 maybe we should document this, or add a `block` flag to `pyRunScript`?

It could look like this: `pyRunScript("test.py" ?block t)`

baptistesion, 2023-06-01:
Okay, thanks for the response! 
Regards,

TM90, 2023-06-01:
> @TM90 maybe we should document this, or add a `block` flag to `pyRunScript`?
> 
> It could look like this: `pyRunScript("test.py" ?block t)`

I think adding the block flag would be a good addition