unihd-cag/skillbridge #221: [SUPPORT] Using the plot function with OCEAN scripting in Python

PaulKokhanov1, 2023-06-22:
Hello, 
I was having issues on how to use the plot function in OCEAN using skillbridge. I am using the code line 
[image]
However I get the following error when running my code
[image]
What would be the best way to solve this issue, as trying to do something along the lines of ws['plot']("VS('/net1')") causes python to think I am attempting to plot the expression "VS('/net1')" instead of just net1, and thus virtuoso is unable to find the expression in my dc results. Any suggestions? Thanks

nielsbuwen, 2023-06-22:
Hi,

can you show me the original OCEAN code that you want to translate?

Is it `plot("VS('/net')")` or `plot(VS('/net'))`.

The first code would be `ws['plot']("VS('/net')")` in python and the second is a bit more complicated:

```
ws['plot'](ws['VS']('/net'))  # this should work
```

PaulKokhanov1, 2023-06-23:
Hello,

Yes it was the second one, specifically,
[image]

But `ws['plot'](ws['VS']('/net'))` did work! Thank you so much.

nielsbuwen, 2023-06-23:
Glad i could help.

Depending on the size of the return value of `VS("/net")` it could be faster to only compute that part on the SKILL side.

```python
ws['plot'](ws['VS'].var("/net"))  # notice the 'var'
```

`ws['VS'].var` is a symbolic expression that is only evaluated on the SKILL side. My original answer sends the results from `VS` to python and then back to SKILL. With the new snippet you save that round-trip. The downside is that you can't use the result from `VS` on the python side.

Feel free to close the issue, when you don't have any more questions about this.

closed 2023-06-23 by PaulKokhanov1