unihd-cag/skillbridge #222: [SUPPORT] defining output path for Spectre Simulation Results

PaulKokhanov1, 2023-06-26:
Hello, I was wondering how to use skillbridge to define an output path for the virtuoso spectre simulation results. Currently it seems that the outputs are place in the same directory as my design reference except it is creating a new "distributed" folder and placing the netlist and psf results in this folder. I am unsure why it is creating the "distributed" folder, i'm not sure if this is a feature within virtuoso or whether skillbridge implements this. Ideally I would like to be able to define my simulation output path. Would this be done via cdslib? If so, using python how can I define where the script should look for the cdslib script? Thanks
[image]
[image]

TM90, 2023-06-26:
Hello,

Skillbridge is only a thin layout to do type translation and interfacing between skill and python.

To read and change the psf directory paths I would recommend to take a look into the Virtuoso ADE SKILL Reference.
We cannot link cadence documentaiton directly since it is proprietary, so you will need a Cadence Support Account.

I.e. 
``` python
# Skill method:
axlGetResultsLocation(...)
# skillbridge access:
ws.axl.get_results_location(...)
...
```

Hope this helps.

Best regards

Tobias

TM90, 2023-06-28:
I will close the issue for now If you have additional questions feel free to reopen the issue or open a new one.

Best regards
Tobias

closed 2023-06-28 by TM90