unihd-cag/skillbridge #223: Unable to plot current of transient analysis using Skillbridge

PaulKokhanov1, 2023-07-05:
Hello, I am unsure whether I am failing to understand a property of ocean scripting or whether this is a bug within skillbridge. Essentially, whenever I try to save a current using skillbridge through the following command:
`ws['save']( Symbol('i'), 'I0.cout' )`

The results of my simulation will not generate the plot for the current.
Once I run my simulation I open & select the Results and I see the following output:
[image]

So I can see that "I0.cout" was saved, however, when running
`plot(i("I0.cout"))` in the virtuoso CIW I get the following output:

[image]

Clearly we can see that it is voltage being plotted, not current, interestingly enough, all the other outputs in the output list are also voltages, I know by default the save function will save voltage, so I wonder if there is a bug within skillbridge with this function and it is defaulting to saving voltage instead of current? Otherwise, could someone direct me to the correct method in order to save the current for a specific node? Thanks, so much! Feel free to let me know if you need more information.

nielsbuwen, 2023-07-05:
Hi,

i assume you want to call the equivalent of `save(i("IO.cout"))` in python?

That would be:

```python
i = ws['i'].var
ws['save'](i("IO.cout"))
```

The python code you suggested would be equivalent to `save('i, "IO.cout")` in SKILL.

PaulKokhanov1, 2023-07-05:
Just for further information this is for simulating a netlist via the Spectre Simulator within Virtuoso

closed 2023-07-06 by PaulKokhanov1