unihd-cag/skillbridge #226: [SUPPORT] Direct mode sends incompatible skill command

Nadere03, 2023-07-14:
Hello, I am having some troubles with the direct mode. I use the following code snippet and call it with pyRunScript() after loading the file in CIW.

`from skillbridge import Workspace

ws = Workspace.open(direct=True)
ws.ge.get_edit_cell_view()
ws.close()`

This results in this error:

command "skill:geGetEditCellView( )" resulted in error ("eval" 0 t nil ("*Error* eval: unbound variable" skill))

It seems like there is a skill: at the front of the correct command and I can't figure out why. This is with Python 3.9 and Virtuoso running on a Unix server.

Appreciate any help you can provide.

nielsbuwen, 2023-07-14:
Hi,

is this all of the code you are running? Can you please verify that there is no `print`, `sys.stdout.write` or similar **before** you open the Workspace?

The error you are getting can be reproduced with the following code:

```python
from skillbridge import Workspace

print("skill:", end="")
ws = Workspace.open(direct=True)
ws.ge.get_edit_cell_view()
ws.close()
```

Skillbridge itself does not print this kind of string.

After the workspace is open, you may use `print` as much as you want, but before that it is sent to Virtuoso and interpreted as a SKILL command.

Nadere03, 2023-07-14:
Thanks for the response. I used no `print` or `sys.stdout.write`, but I still managed to resolve the issue. It seems like Skillbridge was already installed by the server at a much lower version and I forgot to check the version after pip install. 

As it is working with 1.5, I will close the issue, but I am not sure if that's actually the problem.

closed 2023-07-14 by Nadere03