unihd-cag/skillbridge #227: [SUPPORT]Queries about the package usage

PiyushSaini07, 2023-07-26:
Hi,

I was exploring the feasibility of writing stimuli to a schematic via python and then generating netlist out of that( for example: giving dc value to a vdc source via python). Came across skillbridge in the process.
To understand the general usage of the package I referred to [blog](https://iamanintrovert.github.io/notes/Running-Spectre-Simulation-from-python/) and [docs page](https://unihd-cag.github.io/skillbridge/examples/basic.html#basic).

However, I have some really basic questions about the package:
1. Is SKILL language a prerequisite to using skillbridge?
2. Can I modify the object(resistor/capacitor/voltage sources) properties from skillbridge?
[image] [image]

3. Is it possible to create a config view or an ADE L view out of python code?

Thanks.

nielsbuwen, 2023-07-26:
Hi,

1.
Knowledge about the SKILL functions from the tool you are using **are** required. But no knowledge about advanced techniques from the SKILL language itself are required. That's one reason why skillbridge exists: to hide the "complicated" details of the SKILL language.

But in order to do anything useful you must know which SKILL functions to use. With a support account from https://support.cadence.com you can access the documentation. Most of the SKILL functions for Virtuoso can be found [here](https://support.cadence.com/apex/techpubDocViewerPage?xmlName=skdfref.xml&title=Virtuoso%20Design%20Environment%20SKILL%20Reference%20--%20Graphics%20Editor%20Functions%20-%20geGetEditCellView&hash=pgfId-1027620&c_version=ICADVM20.1&path=skdfref/skdfrefICADVM20.1/chap1.html#pgfId-1027620)

2.
If it can be done from SKILL than it can be done with skillbridge.

Somewhere in the "Edit Object Properties" window at the bottom there should be a checkbox (something like "show CDF names"). If you check that the property names are shown as they are called in SKILL. "Rise time" is probably called `riseTime` in SKILL, but you have to verify that.

Alternatively, with a handle to the instance, you can list the properties as they are called in python with `print(dir(my_instance))`

To actually change the property, you simply assign a new value to it

```python
from skillbridge import Workspace

ws = Workspace.open()
cell_view = ws.ge.get_edit_cell_view()
instance = ws.db.find_any_inst_by_name(cell_view, "drain_volt")
instance.rise_time = 5
```

@TM90 correct me, if i'm wrong

You may have to experiment a bit which type the parameter accepts. Maybe it's a string or an integer. This is handled by the PCELL and we can't predict that behaviour.

3.
To create a new cell view, take a look at the [Cadence Forum](https://community.cadence.com/cadence_technology_forums/f/custom-ic-skill/36448/skill-command-to-create-the-cell-view).

The function in python then look like this: `ws.db.open_cell_view_by_type(...)`. After that you probably have to save it with `dbSave`.

TM90, 2023-07-26:
> To actually change the property, you simply assign a new value to it
> 
> ```python
> from skillbridge import Workspace
> 
> ws = Workspace.open()
> cell_view = ws.ge.get_edit_cell_view()
> instance = ws.db.find_any_inst_by_name(cell_view, "drain_volt")
> instance.rise_time = 5
> ```
> 
> @TM90 correct me, if i'm wrong
> 
> You may have to experiment a bit which type the parameter accepts. Maybe it's a string or an integer. This is handled by the PCELL and we can't predict that behaviour.
> 
> 
> To create a new cell view, take a look at the [Cadence Forum](https://community.cadence.com/cadence_technology_forums/f/custom-ic-skill/36448/skill-command-to-create-the-cell-view).
> 
> The function in python then look like this: `ws.db.open_cell_view_by_type(...)`. After that you probably have to save it with `dbSave`.

Modifying instances may require you to run the CDF callbacks afterwards to update dependent parameters of the changed instance. A code snipped on how to do that can be found at the Cadence Support just search for update CDF callbacks.

nielsbuwen, 2023-07-26:
I assume that these "voltage sources" are instances in the schematic view? If so then yes, you can create those instances with python. The correct function should be ` dbCreateInst` or `dbCreateInstByMasterName`. You can find the documentation for these functions in the link from my first answer.

I don't want to replicate the documentation here, because i am not sure about the copyright rules.

TM90, 2023-07-27:
Other interisting methods to draw wires and pins might be `schCreateWire` and `schCreatePin`

All the methods are documented in different manuals which can be found at the cadence support page.

They can be accessed from the skillbridge following the prefix and caml case conversion scheme:

``` python
from skillbridge import Workspace

ws = Workspace.open()
# dbCreateInst
ws.db.create_inst(...)
# dbCreateInstByMasterName
ws.db.create_inst_by_master_name(...)
# schCreateWire
ws.sch.create_wire(...)
# schCreatePin
ws.sch.create_pin(...)

```

After creating instances wires etc. the cellview must be saved as mentioned above.

PiyushSaini07, 2023-07-28:
Thanks for the help nielsbuwen and TM90. 
I tried adding a voltage source and updating its properties. Basically, could accomplish what I desired.

Closing the issue.

closed 2023-07-28 by PiyushSaini07