unihd-cag/skillbridge #228: [SUPPORT] Starting pyServer without opening Virutoso

PaulKokhanov1, 2023-07-26:
Hello, this is more of a question on the capabilities of skillbridge rather than an issue. I have a task where I use skill code to extract a list of pins, specifically, 
[image]

I was using skillbridge to launch this procedure during my python script and analyze it using python. However, I was wondering what are my options to launch this piece of code without starting virtuoso? Is it even possible? I am a little doubtful since I would specify the library name, cell name and view name in the procedure call and I imagine it needs virtuoso to be open in order to process it. But maybe there is a way to "start" virtuoso without enabling the GUI? Any ideas/advice or even directions on where to look? Thanks. Feel free to further ask me questions if I was not clear enough with my goal.

TM90, 2023-07-26:
Virtuoso is needed since the methods are translated and executed in the skill interpreter

There is a nograph mode for virtuoso so it can be started without a gui:

But then you still have to start the skillbridge server.

You can echo the command into virtuoso:

``` bash
echo "pyStartServer" | virtuoso -nograph &> /dev/null &
python your_client_script.py
```

In your client script you have to wait until the skillbridge server is started. Otherwise you wont get the Workspace to open.

I hope this helps.

PaulKokhanov1, 2023-07-26:
Thanks for the prompt reply! Do I first have to load the python_server.il prior to invoking the pyStartServer? If so, what would be the code to do this via the terminal?

TM90, 2023-07-26:
Yes! I would put the python_server.il into a corresponding cds init or similar which is auto loaded on startup.

PaulKokhanov1, 2023-07-26:
Sorry but I am very unfamiliar with using the terminal, I realize the init will be auto loaded however, I am not sure how to define and create it, and specifically what "startup" you mean? do you mean when the nograph version of virtuoso is started? Thanks again for all the help!

TM90, 2023-07-27:
There are config files which are loaded on virtuoso startup.

Virtuoso will look at startup for a .cdsinit file in your current working path or home path and load it.

nielsbuwen, 2023-07-27:
Maybe we should document this?

TM90, 2023-07-27:
> Maybe we should document this?

You mean that the `load(python_server.il)` could reside in the .cdsinit?

nielsbuwen, 2023-07-27:
The whole process of starting virtuoso without a gui and connecting skillbridge.

PaulKokhanov1, 2023-07-27:
I see, unfortunately within my company I don't have access to editing the .cdsinit file, what other options do I have?

PaulKokhanov1, 2023-07-27:
is it as simple as just calling it .cdsinit ? how do I specify that it should correlate to the virtuoso startup and not some other application?

TM90, 2023-07-27:
It is just a file called `.cdsinit` the load order and that virtuoso looks for that specific file is virtuoso specific behaviour. Other applications should not look for it.

PaulKokhanov1, 2023-07-27:
I understand, so other applications would have an init file but most likely with a different name, ok, thanks again for the help, I will see what I can do and ask anymore questions if I have any.

PaulKokhanov1, 2023-07-27:
Actually, one quick question, what does this part in the command line code do `&> /dev/null &` ?

TM90, 2023-07-27:
`&>` - redirects stdout and stderr of the application in this case to `/dev/null` so it dismisses it.
the `&` afterwards runs the command in background so it does not block your session.

PaulKokhanov1, 2023-07-27:
I get the following error trying to run that command 
[image]

TM90, 2023-07-28:
> I get the following error trying to run that command [image]

The same command runs sucessfully virtuoso in the background on my side

nielsbuwen, 2023-08-01:
@PaulKokhanov1 what kind of Shell are you using? `virtuoso -nograph &> /dev/null &` is a valid bash command

closed 2023-08-28 by TM90