unihd-cag/skillbridge #234: [SUPPORT] Updating pwl sources

PiyushSaini07, 2023-08-17:
Hi,

I was wondering whether I can update a vpwl source from skillbridge automatically using a loop.

I have a code that manually updates the time value pairs of the source:

```
from skillbridge import Workspace
ws = Workspace.open()

voltSupply = ws.db.find_any_inst_by_name("schematic","vpwl_source")
voltSupply.tvpairs = 3    # updates the number of pairs to 3
voltSuppy.t2 = "10n"
voltSuppy.v2 = 1
voltSuppy.t3 = "100n"
voltSuppy.v3 = 4
```

Is there a way to initialize these time value pairs inside a loop?
Or if I can pass them from a list (for example: `[("0n",0),("10n",1),("100n",4)]`).

I looked for vpwlf source, but that requires a file input and I want to avoid an extra file.

nielsbuwen, 2023-08-17:
You can store the attribute names and values for each attribute in a list and assign the attributes in a loop, yes.

```python

attributes = [
    ('t2', '10n'),
    ('v2', 1),
    ('t3', '100n'),
    ('v3', 4),
]

for name, value in attributes:
    setattr(voltSupply, name, value)

```

This is nothing skillbridge specific however, just normal python.

PiyushSaini07, 2023-08-17:
It works like a charm, Thanks for the help!

I'm slightly new to python so had no clue this can be done.

closed 2023-08-17 by PiyushSaini07