unihd-cag/skillbridge #238: [SUPPORT] Creating a custom menu in virtuoso

PiyushSaini97, 2023-09-28:
Hi, 

I've been trying to create a custom menu in virtuoso.
Searched a bit on [cadence's forum](https://community.cadence.com/cadence_technology_forums/f/custom-ic-design/17476/how-to-create-personalized-menu-in-virtuoso-layout-editor#:~:text=be%20something%20like%3A-,procedure,-(AmitAddMenu(_args)) for help.

Found this skill code to create a menu.
[image]
 

I rewrote the same code using skillbridge:
```
from skillbridge import Workspace

ws = Workspace.open(direct=True)

def printA():
    print("Printing from print A")

def printB():
    print("Printing from print B")

def createMenu():
    val1 = ws.hi.create_menu_item("item1","print1",print("A"))
    val2 = ws.hi.create_menu_item("item2","print2",printB())

    pullDownMenu = ws.hi.create_pulldown_menu("pullDown","test_menu",[val1,val2])

ws["deRegUserTriggers"]("schematic",None,createMenu())
```

However, when I run the code, I'm getting following error:
```
pyRunScript "/data/pythonUtilities/src/test.py"
ipc:19
command "hiCreateMenuItem(\"item1\" \"print1\" nil )" resulted in error ("hiCreateMenuItem" 0 t nil ("*Error* hiCreateMenuItem: extra arguments or keyword missing" ("item1" "print1" nil)))
PYTHON: A

PYTHON: Traceback (most recent call last):
  File "/data/pythonUtilities/src/test.py", line 19, in <module>
    ws["deRegUserTriggers"]("schematic",None,createMenu())
  File "/data/pythonUtilities/src/test.py", line 14, in createMenu
    val1 = ws.hi.create_menu_item("item1","print1",print("A"))
  File "/projects/lib/python3.6/site-packages/skillbridge/client/functions.py", line 41, in __call__
    result = self._channel.send(command)
  File "/projects/lib/python3.6/site-packages/skillbridge/client/channel.py", line 58, in send
    return self.decode_response(input())
  File "/projects/lib/python3.6/site-packages/skillbridge/client/channel.py", line 47, in decode_response
    raise RuntimeError(response)
RuntimeError: ("hiCreateMenuItem" 0 t nil ("*Error* hiCreateMenuItem: extra arguments or keyword missing" ("item1" "print1" nil)))
```

Turns out the python function is not recognized by skill and is taken as "nil".

How do I make a python function into "callable" function that can be understood by virtuoso/skill? 

Thanks,
Piyush

PiyushSaini97, 2023-09-29:
> 2. You cannot use python callbacks in a skill function

So, I cant run a python function from skillbridge because the skill cant understand python syntax. Did I understand it correct?

> For simple function calls you can use skillbridge to create the string for you like this: ws.prefix.a_specific_function.lazy(first_arg, second_arg, ...) (note the lazy). This string can then be passed as the ?callback argument to the menu creation function.

I could not get this working. Throws error for `ws.prefix`

nielsbuwen, 2023-09-29:
> So, I cant run a python function from skillbridge because the skill cant understand python syntax. Did I understand it correct?

Correct, at least there is no easy way to do it.

> I could not get this working. Throws error for `ws.prefix`

This was only an example. `ws.prefix.a_specific_function` does not exist. If you show me the actual code you want to run inside the callback, i can help translate it

PiyushSaini97, 2023-09-29:
> from skillbridge import Workspace, Symbol
ws = Workspace.open()
skill_print = ws['printf']
ws.hi.create_menu_item(name=Symbol('MyMenu'), item_text="My Menu Text", callback=skill_print.lazy("Hello SKILL\n"))

Understood. For callback argument, I can ideally call a python file from shell and it should just work fine.
Basically, call a skill function.
For example: 
`ws.hi.create_menu_item(name=Symbol('MyMenu'), item_text="My Menu Text", callback=shell("python3 <pathToFile.py>"))`

nielsbuwen, 2023-09-29:
This has the same problem as your first example. `shell("python3", "path")` will be executed on the python side and crash, because `shell` is not a function in python.

This should work:

```python
skill_shell_call = ws['shell'].lazy("python3", <path to file>)
# contains: 'shell("python3" ...)' as a string

ws.hi.create_menu_item(name=Symbol('MyMenu'), item_text="My Menu Text", callback=skill_shell_call)
```

If you want to use skillbridge function inside the triggered python script, you must use a skillbridge function to run the file:

```python
py_script = ws['pyRunScript'].lazy(<path to python file>)

ws.hi.create_menu_item(name=Symbol('MyMenu'), item_text="My Menu Text", callback=py_script)
```

PiyushSaini97, 2023-09-30:
Thanks for the help.
I was able to add the menu to some extent but it required me to run the script each time I opened the schematic view.

Had to write a skill function nevertheless:
I could not get deRegUserTriggers to work in python.
```
procedure( pullDownMenu()
    hiCreatePulldownMenu(
       'CCSLayoutPulldownMenu
       "test_menu"
       list(
          hiCreateMenuItem( ; list of menu items
             ?name 'test1
             ?itemText "test1"
             ?callback   "pyRunScript \"<pathToFile>.py\""
            )
        ) ; list
    ) ; hiCreatePulldownMenu
) ; procedure
 
procedure( userTrigger(args)
   hiInsertBannerMenu(
        args->window
        pullDownMenu()
        length( hiGetBannerMenus( args->window))
   )
) ; procedure
 

 
;; Trigger for schematic
deRegUserTriggers("schematic" nil nil 'userTrigger)
 

```

nielsbuwen, 2023-10-02:
It can be done, but it is a bit ugly:

```python
from skillbridge import Workspace, Symbol, Var

ws = Workspace.open()

# a string containing skill code to run as the menu callback
callback = ws['pyRunScript'].lazy("path_to_script.py")

# the skill code to create a new pulldown menu
create_menu_entry = ws.hi.create_pulldown_menu.var(
    Symbol('pullDown'),  # 'pullDown
    "test_menu",
    [
        ws.hi.create_menu_item.var(
            name=Symbol('test1'),
            item_text="test1",
            callback=callback,
        )
    ],
)

# this defines a new function called 'userRegisterMyMenu'
ws.define(
    'registerMyMenu',
    ['args'],
    ws.hi.insert_banner_menu.lazy(
        Var('args').window,
        create_menu_entry,
        ws['length'].var(ws.hi.get_banner_menus.var(Var('args').window))
    )
)

ws.de.reg_user_triggers('schematic', None, None, Symbol('userRegisterMyMenu'))
```

PiyushSaini97, 2023-10-10:
Thanks for the help.
It was easier to write skill in this case so went ahead with it.

Closing the issue.

closed 2023-10-10 by PiyushSaini97