unihd-cag/skillbridge #239: [SUPPORT] Running a script from PyQt5 window hangs virtuoso

PiyushSaini97, 2023-10-10:
Hi,

I've created a GUI in PyQt5 on which a button press calls the python skill script updating a schematic.
However, once I run the file, the window hangs on the very first skill command.

Adding the files here:
1. skill script to create a menu in schematic:

```
procedure( pullDownMenu()
    hiCreatePulldownMenu(
       'pullDown
       "test_menu"
       list(
          hiCreateMenuItem( ; list of menu items
             ?name 'test1
             ?itemText "test1"
             ?callback   "shell(\"LD_LIBRARY_PATH=/lib /bin/python3 <pathToGUIFile.py>\")"
            )
        ) ; list
    ) ; hiCreatePulldownMenu
) ; procedure
 
procedure( userTrigger(args)
   hiInsertBannerMenu(
        args->window
        pullDownMenu()
        length( hiGetBannerMenus( args->window))
   )
) ; procedure
 
;; Trigger for schematic
deRegUserTriggers("schematic" nil nil 'userTrigger)
```

2. File to create GUI:

```
  def genSchScript():
      os.system("python <pathToMainScript.py>")  #a different virtual environment of python runs the skill script
      window.close()
  app = QApplication(sys.argv)
  # create a main window
  window= QMainWindow()
  
  # Create a button to open the file dialog
  button = QtWidgets.QPushButton(window)
  button.setText("Run Script")
  button.move(10,50)
  button.clicked.connect(genSchScript)

  # run the window
  window.show()
  sys.exit(app.exec_())
```

3. Main Script:

```
from skillbridge import Workspace
ws = Workspace.open('piyush')

print(ws)   # this works

schcellView = ws.ge.get_edit_cell_view() # **virtuoso hangs here**
print(schcellView)
```

Virtuoso hangs as soon as I call any skill function from python.

On Virtuoso CIW,
I'm starting a server:
`pyStartServer ?id "piyush"`

I'm unable to understand why virtuoso hangs when I call the script via my GUI but runs as intended when I call same command from terminal.

Please help,
Piyush

nielsbuwen, 2023-10-15:
Sorry for the late response. This looks like a complex issue. I have to find time to reproduce this...

But currently i have no idea what could cause this, unfortunately.

TM90, 2023-11-13:
@PiyushSaini97 In order reproduce this issue a minimal example would be needed.

PiyushSaini97, 2023-11-17:
Hi,
Sorry for the delay.
This was resolved by our EDA expert.
Unfortunately, no insights of what caused the error or how it was resolved.
Closing the issue

closed 2023-11-17 by PiyushSaini97