unihd-cag/skillbridge #243: [FEATURE] Schematic Pin List

s-y-wu, 2024-01-09:
## Problem
When exporting pin lists from schematics, I want to use skillbridge to set properties from `pinList's prop`. In SKILL, it looks like this:
```lisp
pinList = schSchemToPinList(lib cell "schematic")  ; pinList->prop is nil as reported in CIW...
pinList->prop->schCvCreationMethod = "Replace"     ; yet this is somehow valid (seems like magic to me)
schPinListToView(lib cell "symbol" pinList 'schPinListToSymbolGen)
```
Specifically, line 2 accesses properties of `pinList->prop` despite it being `nil`. If written using skillbridge, Python errors out
```python
from skillbridge import Workspace

ws = Workspace.open()
lib = "foo"
cell = "bar"
pin_list = ws.sch.schem_to_pin_list(lib, cell, "schematic")

print(pin_list["prop"])
# [None]

setattr(pin_list["prop"], "schCvCreationMethod", "Replace")
# AttributeError: 'list' object has no attribute 'schCvCreationMethod'

ws.sch.pin_list_to_view(lib, cell, "symbol", pin_list, "schPinListToSymbolGen")
```

##  Potential Solution
- Add an encoder for`ws.sch.*_to_pin_list` methods. Have them return a special PinList object instead of a `dict` with two attributes: `prop` and `pins`. 
- Set the list currently stored in `pin_list["pins"]` as `pin_list.pins`
- Set`pin_list.prop` stores another class like PinListProperties that intializes `pin_list.prop.schCvCreationMethod = None` 
- Translate any assignment of `pin_list.prop.schCvCreationMethod = "Replace"` into `pinList->prop->schCvCreationMethod = "Replace"`
- Add a decoder for every `ws.sch.pin_list_to_*` to set pin_list.prop back to the current dictionary with `pin_list["prop"] = [None]` before sending it through the current process.

## Additional Context & Bug Report
In my design automation python scripts, I need to:
1.   Export a schematic cell view to create a symbol cell view
2.  Run this without clicking warning popup windows.

The skill API offers a few ways to accomplish task [1]. My first attempt was:
```python
pin_list = ws.sch.schem_to_pin_list(lib, cell, "schematic")
ws.sch.pin_list_to_symbol(lib, cell, "symbol", pin_list)
```
This worked fine for 7 of my schematics (basic pins, nets, and symbol instances). But once I got to my 8th and largest schematic (many symbol instances), **virtuoso crashed every time**. However, I knew this was a skillbridge specific issues because manually inputting the following code in CIW successfully generated the 8th schematic's symbol **without crashing virtuoso, consistently**.
```
pinList = schSchemToPinList(lib cell "schematic")
schPinListToView(lib cell "symbol" pinList 'schPinListToSymbolGen)
```
I suspected the python `pin_list` was responsible. And indeed, this workaround never crashes virtuoso:
```python
ws.sch.view_to_view(lib, cell, lib, cell, "schematic", "symbol", "schSchematicToPinList", "schPinListToSymbol")
```
However, all of these methods described above do not accomplish task [2], as a popup dialog appears to settle the ambiguity of the pin list property's schCvCreationMethod, which could either be "Modify" or "Replace". The only workaround that accomplished both (1) and (2) was manual function loading with Skill code:
```python
skill_code = """
procedure(mySymbolSave(lib cell)
    let( (pin_list)
        pin_list = schSchemToPinList(lib cell "schematic")
        pin_list->prop->schCvCreationMethod = "Replace"
        schPinListToView(lib cell "symbol" pin_list 'schPinListToSymbolGen)
    ) ; let
) ; procedure
"""
with tempfile.NamedTemporaryFile(mode='w+', suffix=".il") as temp_file:
    temp_file.write(skill_code)
    temp_file.seek(0)
    # seek breaks the race condition and allows loaders to read from the beginning
    ws['load'](temp_file.name)
ws['mySymbolSave'](library, cell)
```
This works, but having more direct solution in python would be great!

TM90, 2024-01-10:
> ## Problem
> 
> When exporting pin lists from schematics, I want to use skillbridge to set properties within `pinList->prop`. In SKILL, it looks like this:
> 
> ```lisp
> pinList = schSchemToPinList(lib cell "schematic")  ; pinList->prop is nil as reported in CIW...
> pinList->prop->schCvCreationMethod = "Replace"     ; yet this is somehow valid (seems like magic to me)
> schPinListToView(lib cell "symbol" pinList 'schPinListToSymbolGen)
> ```
> 
> Specifically, line 2 accesses properties of `pinList->prop` despite it being `nil`. If written using skillbridge, Python errors out
> 
> ```python
> from skillbridge import Workspace
> 
> ws = Workspace.open()
> lib = "foo"
> cell = "bar"
> pin_list = ws.sch.schem_to_pin_list(lib, cell, "schematic")
> 
> print(pin_list["prop"])
> # [None]
> 
> setattr(pin_list["prop"], "schCvCreationMethod", "Replace")
> # AttributeError: 'list' object has no attribute 'schCvCreationMethod'
> 
> ws.sch.pin_list_to_view(lib, cell, "symbol", pin_list, "schPinListToSymbolGen")
> ```

The Problem here is that the types between Python and Skill are not completely equivalent. And sometimes the type translation can be  ambiguous. In this on skill side we have a disembodied property list which on python side is translated as a dictionary. The empty disembodied property list `(nil)` is ambiguous in this case and could be either `[None]`, `[False]`, `[[]]` or `{}` on the python side and all are valid in different circumstances.

To still fill the property you can assign a dictionary which will be translated correclty to a disembodied property list.

``` python
from skillbridge import Workspace
 
ws = Workspace.open()
lib = "foo"
cell = "bar"
pin_list = ws.sch.schem_to_pin_list(lib, cell, "schematic")

pin_list["prop"] =  {"schCvCreationMethod": "Replace"}
ws.sch.pin_list_to_view(lib, cell, "symbol", pin_list, "schPinListToSymbolGen")
```

Please let us know whether this solutions works and resolves your problem

s-y-wu, 2024-01-10:
Yes, I tried your suggestion. It looks like skillbridge could digest the new values in `pin_list["prop"]`, and my first 7 schematics were saving as symbols properly without popup windows, as desired.

In this case, I'd be happy to mark the feature request as resolved. Everything looks good, except for this one little thorn.

The 8th schematic is still crashing virtuoso. Again, running the SKILL code manually in the CIW to save the 8th schematic as a symbol does NOT crash virtuoso, so this is distinctly an issue when skillbridge is involved. However, this issue maybe difficult to replicate and debug.

My Python traceback:
```
  File "/path/to/utils.py, line 239, in save_schematic_as_symbol
    ws.sch.pin_list_to_view(library, cell, "symbol", pin_list, "schPinListToSymbolGen")
  File "/path/to/venv/lib/python3.12/site-packages/skillbridge/client/functions.py", line 42, in __call__
    result = self._channel.send(command)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/path/to/venv/lib/python3.12/site-packages/skillbridge/client/channel.py", line 158, in send
    return self._receive_only()
           ^^^^^^^^^^^^^^^^^^^^
  File "/path/to/ver02/venv/lib/python3.12/site-packages/skillbridge/client/channel.py", line 154, in _receive_only
    return self.decode_response(response)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/projects/radio/python_env/ver02/venv/lib/python3.12/site-packages/skillbridge/client/channel.py", line 40, in decode_response
    status, response = response.split(' ', maxsplit=1)
    ^^^^^^^^^^^^^^^^
ValueError: not enough values to unpack (expected 2, got 1)
```

Here's an excerpt from virtuoso's crash report. Looks like there is a segmentation fault:
```
startdate:Mon Jan 8 15:58:02 2024
crashdate:Wed Jan 10 09:20:23 2024
appname:virtuoso
version:@(#)$CDS: virtuoso version IC23.1-64b 10/31/2023 23:29 (cpgbld22) $
subversion:IC23.1-64b.ISR3.27
mem:1931337728 (total 1621516455936memavail:957566592k (total 1976728152k)
signal:SegFault(11)
pstack:Thread 5 (Thread 0x2ae3d9c6e700 (LWP 199732)):
pstack:#0  0x00002ae3bc8b720d in poll () from libc.so.6
pstack:#1  0x00002ae3b3550022 in _xcb_conn_wait () from libxcb.so.1
pstack:#2  0x00002ae3b3551c6f in xcb_wait_for_event () from libxcb.so.1
pstack:#3  0x00002ae3d3ab23e0 in ?? () from libcdsQt5XcbQpa.so.5
pstack:#4  0x00002ae3b0959f6f in ?? () from libcdsQt5Core.so.5
pstack:#5  0x00002ae3a82d1dd5 in start_thread () from libpthread.so.0
pstack:#6  0x00002ae3bc8c1ead in clone () from libc.so.6
pstack:Thread 4 (Thread 0x2ae3da1c1700 (LWP 199752)):
pstack:#0  0x00002ae3bc8b72cf in ppoll () from libc.so.6
pstack:#1  0x00002ae3b0b978a9 in qt_safe_poll(pollfd*, unsigned long, timespec const*) () from libcdsQt5Core.so.5
pstack:#2  0x00002ae3b0b98d33 in QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from libcdsQt5Core.so.5
pstack:#3  0x00002ae3b0b43f5b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from libcdsQt5Core.so.5
pstack:#4  0x00002ae3b0958f1e in QThread::exec() () from libcdsQt5Core.so.5
pstack:#5  0x00002ae3d3bc82c7 in ?? () from libcdsQt5DBus.so.5
pstack:#6  0x00002ae3b0959f6f in ?? () from libcdsQt5Core.so.5
pstack:#7  0x00002ae3a82d1dd5 in start_thread () from libpthread.so.0
pstack:#8  0x00002ae3bc8c1ead in clone () from libc.so.6
pstack:Thread 3 (Thread 0x2ae3e757a700 (LWP 205294)):
pstack:#0  0x00002ae3bc8b8f73 in select () from libc.so.6
pstack:#1  0x000000000c12c7f7 in NotifierThreadProc ()
pstack:#2  0x00002ae3a82d1dd5 in start_thread () from libpthread.so.0
pstack:#3  0x00002ae3bc8c1ead in clone () from libc.so.6
pstack:Thread 2 (Thread 0x2ae3ea175700 (LWP 213898)):
pstack:#0  0x00002ae3a82d8e3d in nanosleep () from libpthread.so.0
pstack:#1  0x0000000019a36956 in eli::eliRecheckControl::recheckWorker() ()
pstack:#2  0x00002ae3bc30de43 in ?? () from libstdc++.so.6
pstack:#3  0x00002ae3a82d1dd5 in start_thread () from libpthread.so.0
pstack:#4  0x00002ae3bc8c1ead in clone () from libc.so.6
pstack:Thread 1 (Thread 0x2ae3d2aa4640 (LWP 199667)):
pstack:#0  0x00002ae3a82d9179 in waitpid () from libpthread.so.0
pstack:#1  0x00002ae3af65eb30 in voSystem () from libvirtuos_sh.so
pstack:#2  0x00002ae3af653e5a in mopProcessCdsLog () from libvirtuos_sh.so
pstack:#3  0x00002ae3af64bf17 in voDoAbortFuncs () from libvirtuos_sh.so
pstack:#4  0x00002ae3af64c0c9 in voSigAbortInfo () from libvirtuos_sh.so
pstack:#5  <signal handler called>
[... 42 lines cut ...]
pstack:#48 0x00002ae3b21b3bef in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from libcdsQt5Widgets.so.5
pstack:#49 0x00002ae3b21bc3c8 in QApplication::notify(QObject*, QEvent*) () from libcdsQt5Widgets.so.5
pstack:#50 0x00002ae3b0b451fa in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from libcdsQt5Core.so.5
pstack:#51 0x00002ae3b0b989cb in QEventDispatcherUNIXPrivate::activateSocketNotifiers() () from libcdsQt5Core.so.5
pstack:#52 0x00002ae3b0b98dfb in QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from libcdsQt5Core.so.5
pstack:#53 0x00002ae3d3ab346e in ?? () from libcdsQt5XcbQpa.so.5
pstack:#54 0x000000001ca5c823 in hiiFillCIWInPortCB ()
pstack:#55 0x00002ae3b12bd204 in iliBGetc () from libil_sh.so
pstack:#56 0x00002ae3b12bec4a in ilGetCharPort () from libil_sh.so
pstack:#57 0x00002ae3b12353d4 in ilFillBuffer(char*, unsigned long*, int, void*) () from libil_sh.so
pstack:#58 0x00002ae3b123a050 in ilyylex(ILYYSTYPE*, ILYYLTYPE*, void*) () from libil_sh.so
[... cut ...]
pstack:#66 0x0000000016c6f261 in TclCommandAdaptor::mainLoop(int, char**, char const*, std::vector<ctuString, std::allocator<ctuString> > const&, void (*)()) ()
pstack:#67 0x00000000135d4259 in finaleEntry(int, char**, char const*, bool, finaleAppHarness*) ()
pstack:#68 0x000000000bcf4eab in rdeStartMain(int, char**) ()
pstack:#69 0x0000000006d9f3ed in main ()
```

If it's possible to glean any information from the crash report, we can try to investigate further. But if it seems like a dead end, I'll just rely on my workaround and we can close the ticket.

nielsbuwen, 2024-01-11:
So it seems like the large pin list is crashing something. We can avoid sending that by using global variables:

```python
g = ws.globals('my_globals')

# notice the `>>` and the `var`
g.pin_list << ws.sch.schem_to_pin_list.var(lib, cell, "schematic")  # pinList = wsSchSchemToPinList(...)
g.properties << ws['get'].var(g.pin_list, 'prop')  # properties = pinList->prop
ws['putprop'](g.properties, 'Replace', 'schCvCreationMethod')  # prop->schCvCreationMethod = "Replace"
ws.sch.pin_list_to_view(lib, cell, "symbol", g.pin_list, Symbol('schPinListToSymbolGen'))
```

s-y-wu, 2024-01-11:
Yes, this solution saves my 8th schematic without crashing Virtuoso! The Globals concept is a neat feature. Thank you, Niels.

closed 2024-01-11 by s-y-wu