unihd-cag/skillbridge #244: [FEATURE] gdmci, gdmco, and other lowercase Generic Design Management Functions

s-y-wu, 2024-01-10:
## Problem
I want to use SKILL's `gdmci` and `gdmco` in skillbridge, but it seems like the feature is missing.
```python
from skillbridge import Workspace

ws = Workspace()
library = "foo"
namespace = "Library"
gdm_spec_foo = ws.gdm.create_spec(library, None, None, None, namespace)
ws.gdm.co(gdm_spec_foo)
# RuntimeError: ("eval" 0 t nil ("*Error* eval: undefined function" gdmCo))
```

I believe `gdmci` and `gdmco` were missed because they break the camelCase convention on the SKILL side.

They're also missing in the type hints in `workspace.pyi`:
```python
     ...
    class gdm:
        add_spec_to_spec_list: staticmethod
        create_spec: staticmethod
        create_spec_from_DDID: staticmethod
        create_spec_list: staticmethod
        execute: staticmethod
        inspect_spec: staticmethod
        is_spec_id: staticmethod
        next_from_spec_list: staticmethod
        obj_is_created: staticmethod
        removename: staticmethod
        reset_spec_list: staticmethod
        spec_listp: staticmethod
        spec_type: staticmethod
        specp: staticmethod
    class ge:
    ...
```

## Potential Solution
Create exceptions for the camelCase translator for gdm's lowercase functions. Here is the full list[^1]:
| Function Name         |
|-----------------------|
| gdmcancel             |
| gdmci                 |
| gdmco                 |
| gdmdelete             |
| gdmexport             |
| gdmhistory            |
| gdmsetdefver          |
| gdmsetname            |
| gdmstatus             |
| gdmsubmit             |
| gdmupdate             |

[^1]: Cadence Application Infrastructure SKILL Reference (IC23.1)

nielsbuwen, 2024-01-11:
SKILL never fails to surprise me.

I'm glad you already found the workaround. We will discuss if and how we can fix that.

s-y-wu, 2024-01-11:
Word. These functions use lowercase to match IBM's Rational ClearCase[^1], which is a dinosaur of a version control tool that is or was integrated into Cadence Virtuoso[^2].

I _think_ Cadence has moved on to Synchronocity/Dassault DesignSync for version control, but the legacy interface remains in the official Skill API.

In any case, I appreciate your attention on this!

[^1]: https://www.ibm.com/docs/en/rational-clearcase/10.0.1?topic=commands-gdmci
[^2]: https://en.wikipedia.org/wiki/Rational_ClearCase#Integrations

TM90, 2024-01-25:
We would leave this issue as it is:

Calling the methods via the direct method is fine.

``` python
ws["gdmco"](gdm_spec_foo)
```

Bringing these methods in our prefix syntax will need special treatment. There are a lot of other skill methods which then could be treated in a specific way too. 

Our scope is a lightweight cadence skill <-> python bridge so we decided to not treat these methods in a special way (especially since the above approach seems a good alternative).

closed 2024-01-29 by TM90