unihd-cag/skillbridge #250: Server CLI executable

curtisma, 2024-03-08:
**Is your feature request related to a problem? Please describe.**

It would be more reliable to use skillbridge if it didn't require an installation of Python to run.  
Also, for some older operating systems, the default Python version is < 3.8.  For example, even RHEL7 only has Python 3.6 available through the package manager.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

We could use PyInstaller to create an executable that allows the skillbridge server to run without Python being installed seperately.
Then the server startup would turn into a CLI call.

The executable could be built in the GitHub Action.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

There are other tools for building an executable.  
PyInstaller happens to be the one I'm familiar with and have used for many years.  I'm not familiar with other options.

**Additional context**
Add any other context or screenshots about the feature request here.

Then the Client could also be a Python executable so that in production for standard users there is no dependency on the system Python installation.

You wouldn't even need Python installed to run a tool using skillbridge

TM90, 2024-03-11:
We do not want to support end of life python versions since they wont get any sercurity fixes. 

See https://devguide.python.org/versions/

We do not expect the Skillbridge protocol to change. That means building the binary once will remain compatible for a long time.

Using this binary would already be possible with two minimal changes:

1. When calling pyStartServer use the ?python option to call your binary instead of the python interpreter

i.e. `pyStartServer ?python "yourExecutable"` 

2. The now called python_server.py will now get the script path as first argument. This is not expected and has to be handled. Afterwards build an executable from this modified version.

TM90, 2024-03-25:
I would close this issue if something new comes up feel free to open it again

closed 2024-03-25 by TM90